MrChromebox/scripts #204: Platform unrecognized APL/RABBID

QuinnLion, 2022-05-12:
**   Device:  (CORAL)
** Platform: (unrecognized)
**  Fw Type: Full ROM / UEFI
**   Fw Ver: MrChromebox-4.16 (04/09/2022)
**    Fw WP: Disabled
*********************************************************

This is while booted up into linux with the UEFI full ROM.   
The device is RABBID.   I thought I'd see if there was a new ROM update, and saw that its' downloading the 'coral' ROM. cancelled. 
Looked through the scripts and saw that there's a specific one for rabbid.  Is the apollolake/coral  one, one size fits all?  or is this just a bug?
I couldn't find out a way to specify to the script to select a different ROM.

there's some similar issues...
#201,#170,#179

MrChromebox, 2022-05-12:
> I thought I'd see if there was a new ROM update

the script notifies you in the menu if that's the case

> and saw that its' downloading the 'coral' ROM. cancelled.

you should have known not to proceed based on the fact that it showed the platform as unrecognized

> Looked through the scripts and saw that there's a specific one for rabbid. Is the apollolake/coral one, one size fits all? 

you answered your own question, they are specific to each board name in all but a handful of cases

> I couldn't find out a way to specify to the script to select a different ROM.

it doesn't yet you do that, because people would end up with a bunch of bricked devices. there is only one correct firmware image for each device.

please provide the output of `sudo dmidecode -s system-product-name` so I can figure out why it's not being identified properly

QuinnLion, 2022-05-13:
> you should have known not to proceed based on the fact that it showed the platform as unrecognized

It was recognized on the initial flash. It seems that it's got the Coral rom on it anyway.

> please provide the output of sudo dmidecode -s system-product-name so I can figure out why it's not being identified properly

 Device is being reported as Coral
     That's what it was when I flashed it initially, too. I guess I didn't catch that.   It seems to operate fine but may be causing other issues like suspend not working properly, cpu gets to halt, but USB is still on and when attempting to resume, acts like a coldboot.

QuinnLion, 2022-05-14:
Flashed stock backup and ran the script again.

Device: ASUS Chromebook C423 (RABBID)
Platform: Intel ApolloLake
Fw Type: ChromeOS Stock w/RW_LEGACY
Fw Ver: Google_Coral.10068.113.0 (11/18/2020)
Fw WP: Disabled

Is this what its supposed to show?
[image]

MrChromebox, 2022-05-14:
that's exactly what I would expect, so really puzzling what went wrong the first time

closed 2022-05-14 by QuinnLion