MrChromebox/scripts #206: Ubuntu server on ASUS Chromebox 3 needs HDMI always connected?

gmcmicken, 2022-05-23:
Turning a Chromebox into a headless server with docker. All went well until I tried to remove HDMI, box became unresponsive but didn't turn off. Rebooting with the HDMI unplugged, ssh server never comes up but I can ping for a moment before it disappears. 

Everything works normally with HDMI connected. Has anyone seen this with ASUS Chromeboxes?

Thanks

redpoisonv3, 2022-05-27:
have you tried a dummy hdmi plug to see if that fixes it?

gmcmicken, 2022-05-27:
> have you tried a dummy hdmi plug to see if that fixes it?

Yes, and it does. Through testing I was able to determine its either HDMI or USB Keyboard required to boot, and more peculiarly if the system is running and both are disconnected, the system is immediately shut down. Not ACPI shutdown either, there are no log messages at all in Linux when these are disconnected, the board is effectively pulled of power the instant the last of either HDMI or USB are unplugged.

I've used a spare Logitech unifying receiver as a workaround.

redpoisonv3, 2022-05-28:
which linux distro are you running ill see if i can replicate the issue on my hp chrome box, the only way i could see a permanent fix would be to create a virtual monitor within the os it self.

gmcmicken, 2022-05-29:
> which linux distro are you running ill see if i can replicate the issue on my hp chrome box, the only way i could see a permanent fix would be to create a virtual monitor within the os it self.

Ubuntu server 20.04

MrChromebox, 2023-02-04:
not a script issue; closing

closed 2023-02-04 by MrChromebox

CyJaySong, 2024-04-29:
@MrChromebox I have also encountered this issue recently. It seems that I did not have this issue when using version 4.21.1 of Coreboot. I have upgraded to the latest version of Coreboot, but I don't know how to go back to the previous version

CyJaySong, 2024-04-29:
@ferrellsl What is the root cause of this problem? System? Or UEFI?

ferrellsl, 2024-04-29:
> @ferrellsl What is the root cause of this problem? System? Or UEFI?

Not sure of the cause, but a dummy plug should solve the problem and they're less than $1 USD on AliExpress.

CyJaySong, 2024-04-29:
@ferrellsl I want to return the Coreboot V4.21.1 and try again

ferrellsl, 2024-04-29:
> @ferrellsl I want to return the Coreboot V4.21.1 and try again
  The older downloads are here:  https://coreboot.org/downloads.html

CyJaySong, 2024-04-29:
@ferrellsl I have downloaded firmware version 4.21.1, but I don't know how to flash it. This script can only update the latest version estimate

ferrellsl, 2024-04-29:
> @ferrellsl I have downloaded firmware version 4.21.1, but I don't know how to flash it. This script can only update the latest version

Better wait for MrChromebox to respond then.  I'm definitely not the guy who can give you instructions for downgrading.

CyJaySong, 2024-04-29:
@ferrellsl Ok, but, thank you, have a good day😄

CyJaySong, 2024-04-29:
@MrChromebox Hello, thank you for your open-source contribution. Can you guide me on how to downgrade the version?

MrChromebox, 2024-04-29:
@CyJaySong I don't support this at all, but here you go:
https://docs.chrultrabook.com/docs/firmware/manually-flashing.html