MrChromebox/scripts #207: "Error trying read config from the 'mrchromebox.tech/firmware-util.sh' file"

Scottua25, 2022-05-26:
New to me Samsung Chromebox XE300M22
Arrived wiped.
Filpped developer mode switch and let it do it's thing until reaching login screen.
Logged in as chronos
Issued command: cd; curl -LOK mrchromebox.tech/firmware-util.sh
Received the following:

Warning: error trying read config from the 'mrchromebox.tech/firmware-util.sh'
Warning: file
curl: no URL specified!
curl: try 'curl --help' or 'curl --manual' for more information

After that I'm returned to the command prompt.
Where do I go from here? I'm trying to get home assistant OS on here as the end goal.

MrChromebox, 2022-05-26:
you copied/pasted the command? sounds like you typed a zero instead of capital O

Scottua25, 2022-05-26:
Retyped. I don't know how to copy/paste while inside the shell. I typed it several times making sure each time it was correct.

MrChromebox, 2022-05-26:
where are you copying it from? the k is lower case, and only needed if you get an SSL error otherwise.

https://mrchromebox.tech/#fwscript

Scottua25, 2022-05-26:
Just updated my last comment as i wasn't sure when you'd see it. sorry.

MrChromebox, 2022-05-26:
well, it either had an issue downloading the file, or writing it to disk

Scottua25, 2022-05-27:
What are the steps that need to occur in order for the script to function correctly? I know it needs to be downloaded and from what little I know and can figure out from the commands, it also needs to be copied into the usr/local/bin location and set to 755 permissions. Does it need to be run from that location as well? I can probably figure out how to get it to do what it needs to do now that it’s at least been downloaded.

MrChromebox, 2022-05-27:
under ChromeOS, normally the script is downloaded and run from the user home dir. copying to /usr/local/bin is not necessary (nor recommended) unless you get an error (not warning) about not being able to exec a script from user home dir (happens on some dev channel builds). Additional utils needed are downloaded to partition 12 of the root device (in this case, /dev/sda12) because that's the only place they can be executed from

Scottua25, 2022-05-27:
So i guess there's nothing to be done then?

Scottua25, 2022-05-27:
ok. I finally got there. has to let it boot all the way into chrome and "browse as guest". Once there it let me open your site and then copy/paste worked.

closed 2022-05-27 by Scottua25

MrChromebox, 2022-05-27:
> ok. I finally got there. has to let it boot all the way into chrome and "browse as guest". Once there it let me open your site and then copy/paste worked.

that's normally not necessary, but these ancient ChromeOS builds/firmware from 2012/13 certainly have their idiosyncrasies. It's been ages since I tested STUMPY with stock ChromeOS