MrChromebox/scripts #211: "Owner has disabled developer mode"

slandden, 2022-07-11:
WHen I try to enter Developer mode I get "Owner of this device has disabled developer mode", on this eBay-bought laptop (that was sold because it is EOLed by Google), because apparently only Google is the real owner of every device people their hard-earned money for.

It appears this laptop I bought had been Google-locked.

Thanks Our All-Benificient Overlord Google!

On Lenovo x131e.

Didn't see any mention of this issue in the FAQ.

slandden, 2022-07-12:
Once I connected to the internet it de-enrolled.

But I will leave this issue here because the web-site should say something about this.

MrChromebox, 2022-07-12:
what exactly would you like me to say about it? Managed devices may or may not allow access to developer mode. Previously managed devices may or may not be properly de-enrolled before resale. That's not really within the scope of what I do though

slandden, 2022-07-12:
It is just reoate to ypur dev mpde faq

"Google can disable dev mode remotely, and does so xxxxxxxx"
-- 
 Enviado con Tutanota, disfruta del correo seguro y sin publicidad.

MrChromebox, 2022-07-12:
Google doesn't disable it remotely.

closed 2022-07-12 by MrChromebox