MrChromebox/scripts #215: Two buttons not working on x131e

slandden, 2022-07-16:
Both my "fn" button, and the "keyboard" (compose?) key on the right side do not work. If I use `evtest` and press them there is an event, but no keys are pressed or released. Using `evtest` also shows that this keyboard is *not* usb, but PS/2: "AT Translated Set 2 keyboard".

slandden, 2022-07-16:
Because the keyboard is PS/2 I suspect it has to do with the build of coreboot, and that perhaps the keyboard needs SMM.

MrChromebox, 2022-07-16:
coreboot has nothing to do with the ps2 keyboard, that's handled entirely by the (closed source on this older model) EC

slandden, 2022-07-18:
Actually atkbd is geting the keycodes (65 and 66), but doesn't know what to do with them, and it is more than just seting key codes in user space, as both of these buttons do fancy things.

MrChromebox, 2022-09-09:
closing as not a script issue

closed 2022-09-09 by MrChromebox

slandden, 2022-09-09:
> Closed > #215 <https://github.com/MrChromebox/scripts/issues/215>>  as completed.

These devices are working great. Thank you so much fr letting me buy these two working computers for $35 each as google leaves people bricks