MrChromebox/scripts #217: HP Chromebook 14a - doesn't see USB nor SD on Tianocore using RW_Legacy

jxxnew, 2022-07-27:
HP Chromebook 14a (LANTIS/Jasperlake) fails to see SD card and USB stick at boot into alternative bootloader.

Have used both SD and USB to do a chromebook recovery, so I know the drives work at boot time, but going into grub shell and doing ls with no drive, then with SD and USB in turn, shows no new drives.  On one iteration, a (cd0) appeared but ls (cd0)/ showed invalid filesystem.  Normal drives are (hd0,1-12) (hd1) (hd2) and infrequently (cd0)

Have used Kali, Ubuntu. Debian live ISO's using Rufus (dd mode) and etcher, no joy.

Any suggestions?  with LANTIS/Jasperlake, I have no other option except RW_Legacy.

jxxnew, 2022-07-27:
Also, tried to use a Kali live image renamed to .bin as source for recovery USB, no joy.  

Dev_boot_usb = 1
dev_boot_legacy = 1
dev_default_boot = 0
dev_boot_signed_only = 0

USB does light up (access) during loading of Tianocore

Normal boot shows SD and USB as /dev/sda or /dev/sdb and can read contents without issue

MrChromebox, 2022-07-27:
quite simply, it's a Tianocore bug/issue. I'd (or you'd) need an affected JSL device, a Suzy-Q cable, and a debug build of Tianocore to try and diagnose the issue.

> Also, tried to use a Kali live image renamed to .bin as source for recovery USB, no joy.

of course not, Recovery mode only boots a signed ChromeOS image/kernel

jxxnew, 2022-07-30:
I procured an older test chromebook and followed the same process.

HP Chromebook 11 G5 EE - RELM/Braswell

Process worked without any issue on first try.

One difference, the script installed SeaBios, not Tianocore.   Is this expected behavior and if so is there a point where SeaBios is not the best replacement for RW_legacy?   

For instance, if I wanted to get the most recent Chromebook that supported SeaBios and were most compatible without going EUFI,, would that cutoff be Kabylake / Amberlake?  This guess just comes from the compatibility matrix, that seems to be where there is a systemic change between RW_legacy and UEFI.

Thanks for the help.

MrChromebox, 2022-07-30:
it's not a SeaBIOS/Tianocore issue, most boards have working SD card readers regardless of payload used. The changeover is with boards which use the Alternative Bootloader Menu. Some KBL have it, everything forward from there does.

jxxnew, 2022-07-30:
Just so I understand.

If the issue I faced with a newer Chromebook was a Tianocore issue not recognizing USB nor SD at boot and going to an older Chromebook installed SeaBIOS instead of Tianocore for RW_legacy (or so it appears - same script operation but it appears a different payload was delivered).  Are you saying the real payload the same even with different boot messages or did I bypass the Tianocore issue by having SeaBIOS delivered instead?  

In other words, what seems to have changed is a different payload (SeaBIOS) bypassing a Tianocore issue.  The payload difference is triggered by the specific board in the Chromebook, but the problem is not necessarily Tianocore vs. SeaBIOS but I would likely have seen the same problem even if SeaBIOS was installed on the newer one?

Just trying to understand what changed or where the error resides.  If I could get SeaBIOS installed on the newer one, this would not address the issue in your experience?

Thanks again.

MrChromebox, 2022-07-30:
>  Tianocore issue not recognizing USB nor SD at boot

some SD card readers have issues, it depends on the platform/how the SD reader is wired. Every board is different. USB devices will work under Tianocore. Sometimes you have to remove/reinsert from the main menu. Some USB sticks are just temperamental. But it's been tested working on pretty much every single board.

> I would likely have seen the same problem even if SeaBIOS was installed on the newer one?

no, SeaBIOS and Tianocore each have their own drivers/init for boot devices, so a problem on one tells you nothing about how the other might behave on the same hardware

> If I could get SeaBIOS installed on the newer one, this would not address the issue in your experience?

If SeaBIOS booted on newer platforms I'd offer it. It doesn't. it's 2022, Intel stopped supporting Legacy BIOS/legacy boot years ago. And I don't have the time/interest to try and make it work

jxxnew, 2022-07-30:
Thanks, makes sense.

Appreciate the detailed explanation.

MrChromebox, 2022-07-30:
so I recommend retrying on LANTIS with simply removing/reinserting the USB/SD on the main menu, then going into the boot menu

jxxnew, 2022-07-30:
I'll give that a try

jxxnew, 2022-07-30:
Test process:

1. Start with known good SD and USB with Kali live image.
2. Boot new Chromebook to Chrome OS
3. Insert each external device
4. Result - ChromeOS sees them both
5. Reboot, leaving both devices in
6. Choose Alternative Boot Load
7. Go to grubb command line
8. ls shows internal device with partitions, three other devices. none of which has a recognizable filesystem (note, with no device, hd0, hd1, hd2 shown, with both devices plugged in, cd0 is added)  No joy booting
9. Reboot to ChromeOS with both devices
10. Result - ChromeOS freezes at the splash screen
11. Remove one device at a time and repeat - get to boot menu, ChromeOS normal boot doesn't like both devices (which are both bootable) inserted at same time.
12. Repeat process with each device
13. No joy with SD card - unplug and replug before going into boot menu - no change.
14. No joy with USB - unplug and replug before going into boot menu - no change.

So if Grub is supposed to autoconfig, that is not happening.  ls from grubb command line never shows any new devices with either plugged in at boot, or unplugged and replugged at first screen.  But if boot to internal drive, both devices are seen and fully accessible.

jxxnew, 2022-07-30:
Note: known good means they boot just fine with the older Chromebook.

MrChromebox, 2022-07-30:
> Choose Alternative Boot Load  
> Go to grubb command line

ok, not sure why you're doing that -- that's not how you boot. You need to press ESC when prompted, then go into the Tianocore UEFI menu. Insert the USB/SD, wait a sec, then go into the Boot Menu.

no idea what steps 9-11 are supposed to show, my RW_LEGACY firmware has nothing to do with how ChromeOS behaves.

MrChromebox, 2022-07-30:
also, a USB/SD device which boots via Legacy BIOS (SeaBIOS) has no bearing on it booting via Tianocore (UEFI). Linux ISOs should be written as disc images (dd mode).

jxxnew, 2022-07-30:
After using the firmware script, when the Chromebook boots, it goes to a screen saying you are in developer mode - there is no prompt for ESC, that is what I see in the older Chromebook with SeaBios.

So it boots to screen with:

Return to secure mode
Boot from internal disk
Boot from external disk
Select alternate bootloader

That's with pressing no keys.

Go into Select Alternate Bootloader
Then select Tianocore
Get Tianocore screen with Esc for boot settings prompt

No action - then you get error: file /syslinux/vmlinuz.A not found, 
then get dumped to Grub screen with 5 options - 

local image a, 
local image b, 
verified image A, 
verified image B, 
Alternate USB boot.

choose Alternate USB boot, get errorL file /boot/vmlinuz not found

So I go into the grub command line to see what it can see - and all it sees is the local disk and filesystems. Hence my comments.

It always goes to these same menus with these same errors, with both USB and SD inserted before startup, or reinserted at first menu or after verified valid and accessible in ChromeOS and reboot. 

Grubb edit shows it is looking at hd0,3 (local disk) for (hd0.3)/boot/vminuz  for the alternate USB Boot.

The other detail is just the process and iterations I did in detail. 

[image]

jxxnew, 2022-07-30:
After choosing Tianocore, get a brief prompt for Esc.

Get an HP Lantis Boot screen

Boot Menu shows two devices
   eMMC device - choose this, dumps to Grub with same failures above.
   UEFI shell - This drops me to a shell, exit takes me back to Boot menu

Go back, pick Boot Manager

Change boot order - eMMC or UEFI Shell - if I change the order, then commit, the commit is not persistent
Add boot option - just shows 4 options mapped to the grubb Hd0 disk - partitions 3,5,8,12)
Boot from a file - same options

Note, none of these options change whether or not an SD or USB is present, removed and reinserted, or absent.

So no joy if I hit ESC on Tianocore or just let it go to failure

Looks like the SD and USB is never seen by

jxxnew, 2022-07-30:
on the previous comment, both USB and SD devices are ISO images written using DD and work successfully with no alterations using the older Chromebook and SeaBIOS.

MrChromebox, 2022-07-31:
> No action - then you get error: file /syslinux/vmlinuz.A not found,
> then get dumped to Grub screen with 5 options -

This is a remnant of ChromiumOS, DON'T USE THIS

> After choosing Tianocore, get a brief prompt for Esc.

* press ESC
* wait for settings menu
* insert USB/SD
* got to boot menu
* select USB/SD to boot from

THAT'S IT. don't mess with anything else or any other boot options.

jxxnew, 2022-07-31:
Will do.

Thanks, as always, for the help.

jxxnew, 2022-07-31:
Ran the test.

USB successful when inserted at Tianocore menu, then into Boot menu.   SD card, no such luck.  With the USB, you can see the drive being accessed via lights on reinsert, but with SD of course, no indication that a new device discovery is happening.

But this is a great improvement.

Thanks for making it more clear.

MrChromebox, 2022-07-31:
I'd need a debug log to figure out if it's a simple timeout issue with the SD card or something else. Only way to do that is with a debug tianocore build + debug stock firmware + Suzy-Q cable

jxxnew, 2022-08-01:
I ran another test, using a USB hub that had an embedded microSD card reader.

Worked like a charm.

I'd be glad to help and can probably get a Suzy-Q cable if they are still available (not on Amazon that I can find), but I'm not a developer so the diagnostic process may be a bit beyond my experience.

MrChromebox, 2023-02-04:
closing as not a script issue

closed 2023-02-04 by MrChromebox

dunk3rs420, 2025-01-31:
So I had issues following all info and guide up till this point with chromebook. Once I select my boot usb, it instantly freezes at "WELCOME TO GRUB"
and does not proceed with boot from USB. No idea how to fix this. Thanks.