MrChromebox/scripts #218: Acer Spin 713 - i5 Tiger Lake - Does nothing once Tianocore is selected, and then I need to hard reboot

rainingcrypto, 2022-08-05:
As the title suggests, after a fresh install of MrChromebox's firmware utility (legacy RW), and after a reboot and selecting Tianocore, the system freezes (still shows the screen of the Dev mode, with Tiano highlighted). However, I can't do anything and need to reboot.
Please help! 

Thank you!

MrChromebox, 2022-08-05:
not much I can do about RW_LEGACY failing to boot without the device in hand, as there's no way to debug without a special debug build of the stock ChromeOS firmware and a Suzy-Q cable

LukeShortCloud, 2022-08-10:
I have ran into the same exact issue with the same Chromebook. I saw [here](https://github.com/MrChromebox/scripts/issues/193#issuecomment-1098342971) that you may need this information (although the legacy firmware applies fine without any errors so this may be irrelevant):

Board: `volteer`
Hardware ID: `VOXEL-GFMQ D3B-B2B-E5E-49A-C45`

The SuzyQable appears to be [unavailable](https://www.sparkfun.com/products/retired/14746) now but, if I could get my hands on one, I'd be happy to help debug this.

rainingcrypto, 2022-08-10:
@MrChromebox, what should Luke and I do to get this fixed / debugged? If we get our hands on a Suzy Qable - then what do we do?

MrChromebox, 2022-08-11:
nothing, it's not debuggable without a debug/serial-enabled stock firmware build, which you'd have to get from someone @ Google.

closed 2022-09-26 by MrChromebox