MrChromebox/scripts #219: Pixelbook 2017 downgrade touchpad firmware error despite

gibwise, 2022-08-06:
I have take off battery to disable the write protection. And I stll cannot downgrade the touchpad firmware.
sudo ./flashrom -p host -p ec:type=tp -i EC_RW -w -o rose_v1.1.8546-ee1861e9e.bin
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 4.16.18-galliumos (x86_64)
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 4.16.18-galliumos (x86_64)
Calibrating delay loop... OK.
coreboot table found at 0x7aa99000.
Found chipset "Intel Skylake". Enabling flash write... WARNING: SPI Configuration Lockdown activated.
OK.
Error: Programmer initialization failed.

flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 4.16.18-galliumos (x86_64)
flashrom was built with unknown PCI library, GCC 7.5.0, big endian
Command line (10 args): /tmp/flashrom -p host -p ec:type=tp -i EC_RW -w -o /tmp/flashrom.log rose_v1.1.8546-ee1861e9e.bin
Acquiring lock (timeout=180 sec)...
Opened file lock "/run/lock/firmware_utility_lock"
Lock acquired.
disable_power_management: Disabling power management.
Calibrating delay loop... OS timer resolution is 1 usecs, 3181M loops per second, 10 myus = 10 us, 100 myus = 100 us, 1000 myus = 998 us, 10000 myus = 9981 us, 4 myus = 4 us, OK.
Initializing internal programmer
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-000004d0
Found coreboot table at 0x00000000.
coreboot table found at 0x7aa99000.
coreboot header(24) checksum: d914 table(1208) checksum: 905d entries: 40
Vendor ID: Google, part ID: Eve
Using External DMI decoder.
DMI string chassis-type: "Convertible"
DMI chassis-type is not specific enough.
DMI string system-manufacturer: "Google"
DMI string system-product-name: "Eve"
DMI string system-version: "1.0"
DMI string baseboard-manufacturer: "Google"
DMI string baseboard-product-name: "Eve"
DMI string baseboard-version: "1.0"
get_target_bus_from_chipset() returns 0x10.
Found chipset "Intel Skylake" with PCI ID 8086:9d24. Enabling flash write... 
BIOS_CNTL = 0x8b: BIOS Lock Enable: enabled, BIOS Write Enable: enabled
SPI BAR is = 0xfe010000
GCS = 0x8b: BIOS Interface Lock-Down: enabled, Boot BIOS Straps: 0x0 (SPI)
SPIBAR = 0x00007f7702ffa000 + 0x0000
ich_ generation 16
0x04: 0x3f00f800 (HSFSC)
WARNING: SPI Configuration Lockdown activated.
0x08: 0x00bb1fc0 (FADDR)
0x50: 0x00004acb (FRAP)
BMWAG 0x00, BMRAG 0x00, BRWA 0x4a, BRRA 0xcb
0x54: 0x00000000 (FREG0: Flash Descriptor)
0x00000000-0x00000fff is read-only
0x58: 0x0fff0200 (FREG1: BIOS)
0x00200000-0x00ffffff is read-write
0x5C: 0x01ff0001 (FREG2: Management Engine)
0x00001000-0x001fffff is locked
0x60: 0x00007fff (FREG3: Gigabit Ethernet)
Gigabit Ethernet region is unused.
0x64: 0x00007fff (FREG4: Platform Data)
Platform Data region is unused.
0x68: 0x00007fff (FREG5: Device Expansion)
Device Expansion region is unused.
0x6C: 0x00007fff (FREG6: Reserved 1)
Reserved 1 region is unused.
0x70: 0x00007fff (FREG7: Reserved 2)
Reserved 2 region is unused.
0x74: 0x00007fff (FREG8: Embedded Controller)
Reading flash descriptors mapped by the chipset via FDOC/FDOD... done.
Embedded Controller region is unused.
0x84: 0x00000000 (PR0, unused)
0x88: 0x00000000 (PR1, unused)
0x8C: 0x00000000 (PR2, unused)
0x90: 0x00000000 (PR3, unused)
0x94: 0x00000000 (PR4, unused)
0x98: 0x00000000 (PR5, unused)
0x9C: 0x00000000 (PR6, unused)
0xA0: 0xfe000080 (PR7)
0x00080000-0x01e00fff is read-only
0xA4: 0xb32d5006 (PR8)
0x01006000-0x0132dfff is read-only
Reading flash descriptors mapped by the chipset via FDOC/FDOD... done.
=== Content Section ===
FLVALSIG 0x0ff0a55a
FLMAP0   0x00040003
FLMAP1   0x42100208
FLMAP2   0x00310330

--- Details ---
NR          (Number of Regions):                     1
FRBA        (Flash Region Base Address):         0x040
NC          (Number of Components):                  1
FCBA        (Flash Component Base Address):      0x030
ISL         (ICH/PCH Strap Length):                 66
FISBA/FPSBA (Flash ICH/PCH Strap Base Address):  0x100
NM          (Number of Masters):                     3
FMBA        (Flash Master Base Address):         0x080
MSL/PSL     (MCH/PROC Strap Length):                 3
FMSBA       (Flash MCH/PROC Strap Base Address): 0x300

=== Component Section ===
FLCOMP   0x125c00f5
FLILL    0xad604221

--- Details ---
Component 1 density:            16 MB
Component 2 is not used.
Read Clock Frequency:           reserved
Read ID and Status Clock Freq.: reserved
Write and Erase Clock Freq.:    reserved
Fast Read is supported.
Fast Read Clock Frequency:      reserved
Invalid instruction 0:          0x21
Invalid instruction 1:          0x42
Invalid instruction 2:          0x60
Invalid instruction 3:          0xad

=== Region Section ===
FLREG0   0x00000000
FLREG1   0x0fff0200

--- Details ---
Region 0 (Descr.) 0x00000000 - 0x00000fff
Region 1 (BIOS  ) 0x00200000 - 0x00ffffff

prettyprint_ich_descriptor_master: cs=16
=== Master Section ===
FLMSTR1  0x00a00b00
FLMSTR2  0x00c00d00
FLMSTR3  0x00800800
FLMSTR4  0x00000000
FLMSTR5  0x10010100

--- Details ---
      Descr. BIOS ME GbE Plat EC
BIOS    r     rw      rw        
ME      r         rw  rw        
GbE                   rw        
Plat                            
EC      r                     rw

OK.
Target tp used
cros_ec_probe_dev: probing for CROS_EC at /dev/cros_tp
cros_ec_probe_lpc():518 ...
Target tp used
cros_ec_lpc only supports "ec" type devices.
mec1308 only supports "ec" type devices
mec1308_probe_spi_flash(): entered
mec1308 only supports "ec" type devices
ene_probe_spi_flash
ene_lpc only supports "ec" type devices
Error: Programmer initialization failed.
Restoring PCI config space for 00:1f:5 reg 0xdc
restore_power_management: Re-enabling power management.

MrChromebox, 2022-08-07:
pushed a fix, LMK if it works since I can't test. Must be flashed from CrOS

gibwise, 2022-08-25:
> pushed a fix, LMK if it works since I can't test. Must be flashed from CrOS

I didn`t backup original BIOS so I cannot go back to Chrome OS. Is it possible the touchpad firmware to flash using SuzzyQ cable?

MrChromebox, 2022-08-25:
you can restore the stock firmware using my script, it will provide a firmware image for EVE.

You can almost certainly flash the TP firmware via a Suzy-Q cable, but I haven't done it and don't have any instructions on how to do it. I suspect there's something on chromium.org somewhere