MrChromebox/scripts #220: Required package 'dmidecode' not found even after installing it

51-25, 2022-08-07:
Hello, my chromebook is currently running arch linux (this is my first arch install and was also just installed). when I tried to run the firmware script I got the error `Required package 'dmidecode' not found; cannot continue.  Please install and try again.` so obviously I installed `dmidecode` with the command `sudo pacman -S dmidecode` it installed version 3.4-1. Dmidecode runs fine but only with `sudo` which I suspect is normal. How can I fix this error as I can see that my bios version is 2+ years old?

Some info about the chromebook as gathered from dmidecode is:

```
BIOS Information
	Vendor: coreboot
	Version: MrChromebox-4.12
	Release Date: 06/04/2020
	ROM Size: 8 MB
	Characteristics:
		PCI is supported
		PC Card (PCMCIA) is supported
		BIOS is upgradeable
		Selectable boot is supported
		ACPI is supported
		Targeted content distribution is supported
	BIOS Revision: 4.12

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: GOOGLE
	Product Name: Gnawty
	Version: 1.0
	Serial Number: 123456789
	UUID: Not Settable
	Wake-up Type: Reserved
	SKU Number: Not Specified
	Family: Not Specified
```

MrChromebox, 2022-08-07:
I'd boot a Debian live USB and just update from there. Too many variables to try and debug in Arch

KELP1234, 2022-08-08:
Hello again I have found out what was going on! I spent the last three hours trying different stuff in this order:

on Arch Linux
- downgrading dmidecode to 3.2-1 from 3.4-1
```
Unable to read current firmware; cannot continue.
Either add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu).
See https://www.flashrom.org/FAQ for more info.
```
- added `iomem=relaxed` to my rEFInd config, rebooted and tried again `started to work`
- upgrading from 3.2-1 to 3.3-1 `still worked`

switch to a GalliumOs live usb
- dmidecode 3.1-1 `worked`
- update dmidecode to 3.4-1 _`still worked despite not working on arch`_

Back on Arch Linux
- updated dmidecode from 3.3-1 to 3.4-1 _`worked?`_

closed 2022-08-08 by KELP1234