MrChromebox/scripts #224: Write Protect problem (EVE)

EMLommers, 2022-09-02:
Hi Matt,

I have a problem with EVE and Babymega, after flashing 4.16, officially I cannot upgrade to 4.17

problems that occur:
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 5.19.4-200.fc36.x86_64 (x86_64)
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 5.19.4-200.fc36.x86_64 (x86_64)
coreboot table found at 0x7aa99000.
Found chipset "Intel Skylake". Enabling flash write... WARNING: SPI Configuration Lockdown activated.
OK.
Transaction error between offset 0x00000000 and 0x00000000 (= 0x00000000 + 0)
Writing Status register failed
!!w25_disable_writeprotect(): error=1.
FAILED

WP: status: 0xc0
WP: status.srp0: 1
WP: write protect is enabled.
WP: write protect range:0 - 0

However if I download the .rom file and  flashrom -w coreboot.18072022.rom it gives no error and shows 4.17
The normal way mentioned WP enabled and not possible to upgrade from 4.16 to 4.17.2.

CCD is on factory and all 1's

What could cause the problem?

MrChromebox, 2022-09-02:
sounds like an issue with the version of flashrom the script is using (which is **not** the same as in the unbricking guide) and your OS/kernel

EMLommers, 2022-09-08:
Tested with running kernes latest official.
Babamega on debían an eve on fedora.
For flashing maybe u have to use kernel <= 5.4 or just Ubuntu and fedora 2 releases back 

Also off course I found a workaround, the babamega with APL after flashing and closing laptop I encountered I guess an S3 error. It won't come back after opening laptop and just reboots.

That's it
Thanks

EMLommers, 2022-09-08:
I have tried rebuilding flashrom from third-party, umbrick, hcdtools and standard your flashrom...
All the same and all above errors. Will try with Deb 20.4
Get back
I will check with.

EMLommers, 2022-09-09:
Running Debian on BABYMEGA (5.15.46) and Fedora (5.19.6-200) on EVE.
Standard using the latest kernels...
None of them work.... However... with upgrading or ....
Tried all kind of configs and chromiumos flashrom I guess max works on 5.4

Running Debian 18.04.5/6 and mrchromebox sciript on BABYMEGA works!  Have to keep in mind to use 18.04

On EVE, even running with kernel 5.4 get same results. Error W25!
On EVE, running debian 18.04.6 same results.. Error: W25!

Protection 0xc0

OK.
Transaction error between offset 0x00000000 and 0x00000000 (= 0x00000000 + 0)
Writing Status register failed
!!w25_disable_writeprotect(): error=1.
FAILED
 
Have CCD configured on Testlab and all Y default =1

What am I missing??

MrChromebox, 2022-09-09:
I have no problem running my script / flashing the firmware on any supported models with kernel 5.15/5.18, including REEF, which is the same platform/baseboard as BABYMEGA. 

No idea what's different with your device/config but this isn't a widespread issue

EMLommers, 2022-09-09:
BABYMEGA is solved..... it is working and updated.. SUPER!

Problem  now only is with EVE.

MrChromebox, 2022-09-09:
using an old/unsupported kernel isn't solving the problem

EMLommers, 2022-09-09:
I will just wait and dont compain..... 

An idea for you: why cant users submit their email in order to get news or notifications about new firmware..
You gain emailadresses and further it is up to you.....
Also I can write a script to check for firmware updates.... 
However option 1 is more interesting for you....

EMLommers, 2022-09-10:
ve
If using an old/unsupported kernel isn't solving the problem, what will solve the problem?
I really have no clue! (for EVE) (the solution II have is to download your new rom and update the vpd, and flash, even there is a wp enabled, but works, that's the solution i used to update to 4.17)

MrChromebox, 2022-09-10:
honestly not sure what the problem is, and I'd need to have a full flashrom log of the issue to try and debug it. But since no one else is having this issue, it seems specific to your device/setup.

Do me a favor and boot Pop_OS! 22.04 live USB and run my script from there on both devices, and LMK if either has an issue. If so, then pastebin the time/datestampped flashrom.log in /tmp and post the URL here

EMLommers, 2022-09-17:
Flashing Babymega (Coral) is ok, even with Pop OS 22.04.
Flashrom log: https://pastebin.com/7exLjTE9

Eve will follow shorty..

EMLommers, 2022-09-17:
Flashing with Pop_Os 22.04 gives WP Enabled.
Trying to flash option 1, in order to get log file, only generates message : 
Hardware write-protect enabled, cannot flash Full ROM firmware.

dmidecode shows:
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.
Table at 0x7AA42000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: coreboot
	Version: MrChromebox-4.17.2
	Release Date: 07/18/2022
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PC Card (PCMCIA) is supported
		BIOS is upgradeable
		Selectable boot is supported
		ACPI is supported
		Targeted content distribution is supported
	BIOS Revision: 4.17
	Firmware Revision: 0.0

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Google
	Product Name: Eve
	Version: 1.0
	Serial Number: Not Specified
	UUID: Not Settable
	Wake-up Type: Reserved
	SKU Number: Not Specified
	Family: Google_Eve

Handle 0x0002, DMI type 2, 14 bytes
Base Board Information
	Manufacturer: Google
	Product Name: Eve
	Version: 1.0
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Features: None
	Location In Chassis: Not Specified
	Chassis Handle: 0x0003
	Type: Motherboard

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: Google
	Type: Convertible
	Lock: Not Present
	Version: Not Specified
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: None
	OEM Information: 0x00000000
	Height: Unspecified
	Number Of Power Cords: 1
	Contained Elements: 0
	SKU Number: Not Specified

Handle 0x0004, DMI type 4, 48 bytes
Processor Information
	Socket Designation: CPU0
[... 210 lines cut ...]
Handle 0x0010, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: VGA compatible controller
	Type: Video
	Status: Enabled
	Type Instance: 0
	Bus Address: 0000:00:02.0

Handle 0x0011, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Multimedia audio controller
	Type: Sound
	Status: Enabled
	Type Instance: 0
	Bus Address: 0000:00:1f.3

Handle 0x0012, DMI type 133, 5 bytes
OEM-specific Type
	Header and Data:
		85 05 12 00 01
	Strings:
		KHOIHGIUCCHHII

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table

EMLommers, 2022-09-17:
Flashrom status report:

$ sudo ./flashrom --wp-status
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 5.19.0-76051900-generic (x86_64)
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 5.19.0-76051900-generic (x86_64)
coreboot table found at 0x7aa99000.
Found chipset "Intel Skylake". Enabling flash write... WARNING: SPI Configuration Lockdown activated.
OK.
WP: status: 0xc0
WP: status.srp0: 1
WP: write protect is enabled.
WP: write protect range: start=0x00000000, len=0x00000000

MrChromebox, 2022-09-18:
if flashrom says **software** WP is enabled, then that's what it is.

EMLommers, 2022-10-01:
How it is possible, I have no clue, for Babymega, but in the end I removed battery.....
I noticed you have no stockrom for babymega, to help you see attached. 
Also every recovery image is supplying fw for ro, rw and touch.
Fortunately my EVE right USB is unstable, and had to order a new module, and will open EVE and remove protected as soon as arrived.
Opening EVE is such a time consuming event...;) Prefered not to opened it.

Attached are backups from BABYMEGA and EVE. (Stock roms)
[stock-firmware-EVE-20220506.zip](https://github.com/MrChromebox/scripts/files/9690770/stock-firmware-EVE-20220506.zip)
[stock-firmware-BABYMEGA.zip](https://github.com/MrChromebox/scripts/files/9690771/stock-firmware-BABYMEGA.zip)

Just close the topic!

EMLommers, 2022-10-01:
Also, I found in linux sources Google is checking the version of firmware.
They check if version firmware prefix is 'Google_' , related to ec and other functionality.  If version would be Google_MrChromebox.xxx It could resolve some issues in my opinion
Also would it help also to backup ChromeOS apci rules and incorporate them in your Firmware?

EMLommers, 2022-10-01:
HOWTO:  download a recovery image, rename .bin to .iso, mount ROOT-A.
copy ROOT-A/usr/sbin/chromeos-firmwareupdate to an local rewrtite location.
rename the file to chromeos-firmwareupdate-${chromeos-family-name}, for example chromeos-firmwareupdate-coral, and add .zip extension
Now extract the file and you have the ro/rw/ec (images) for all the family models and sometimes RootKeys / Vblock-A keys.

You are welcome...

EMLommers, 2022-10-01:
After reading you can close this topic! The latest flashrom btw gives more info about pins and programmer is selectable -p ec or host or dediprog, also info about rebooting instead of shutdown in relation to remove protection and w25.

MrChromebox, 2022-10-02:
> Google_MrChromebox.xxx It could resolve some issues in my opinion

I'm not going to mis-label my firmware because Linux's DMI checker is broken. Additionally, there are cases where Google's firmware needs special handling and mine does not, so it is best not to conflate the two. All coreboot firmware is not alike.

> Also would it help also to backup ChromeOS apci rules and incorporate them in your Firmware? Google uses acpi rules a lot!

I don't know what you mean by this.

> PS: download https://dl.google.com/dl/edgedl/chromeos/recovery/recovery.conf
filter unique zip files, extract and get all firmware files you have ever could imagined.

yeah, I know, I wrote the script in coreboot to do this automatically. Doesn't mean I want the responsibility of providing up-to-date stock firmware images for every Chromebook out there.

> Use dmi_sys_vendor as, in capitals, 'GOOGLE'

that is a bug in those drivers, I have submitted patches to fix but have gone nowhere.

closed 2022-10-06 by EMLommers

EMLommers, 2022-10-06:
Ok, clear. 
Just mentioned because, I was lazy, and wanted to use the ChromeOS firmware again to test some stuff, and noticed the notification of Babymega stock firmware was not available, using your script. And had to use the dump or ChromeOS stock firmware.
I will modify some drivers with your DMI firmware mods. 
Also overriding your ACPi tables with stock ChromeOS tables like NHLT and some other.
My determination is to get sound working for all boards. Even SOF,  (or some developers) ,has/have conflict of interest. 
But I think you knew already.
Closed!

MrChromebox, 2022-10-06:
I don't understand what ACPI tables you are saying are different, nothing related to audio should be.

my patches for the various EC drivers simply change the DMI check :

    .matches = {
    DMI_MATCH(DMI_BIOS_VENDOR, "coreboot"),
    DMI_MATCH(DMI_SYS_VENDOR, "GOOGLE"),
    },
    },
    {
    .matches = {
    DMI_MATCH(DMI_BIOS_VENDOR, "coreboot"),
    DMI_MATCH(DMI_SYS_VENDOR, "Google"),
    },
    },

that's all that's needed in any of these drivers for proper matching

EMLommers, 2022-10-06:
Also I will use in cros_ec_lpc.c
DMI_MATCH(DMI_BIOS_VERSION, "Google_",),
But
DMI_MATCH(DMI_BIOS_VERSION, "MrChrom",),
instead.

MrChromebox, 2022-10-06:
that is not the correct way to match, since it doesn't match upstream coreboot builds. The BIOS_VERSION is irrelevant. Matching BIOS_VENDOR and SYS_VENDOR is correct for all Chromebooks running any coreboot firmware

EMLommers, 2022-10-10:
Ok.. Just wanted to help improving your firmware. Than it is irrelevant in  cros_ec_lpc.c and chromeos_pstore.c to check BIOS_VERSION.