MrChromebox/scripts #228: CTRL + L doesn't open legacy boot mode

newmonksh, 2022-10-08:
Installed the script
[image]
[image]

But on reboot pressing CTRL + L shows tow options

[image]

but no other option  Legacy Boot Mode or SeaBIOS

Any suggestions

MrChromebox, 2022-10-08:
that is the expected result on newer Chromebooks. What's the problem?

newmonksh, 2022-10-08:
Thanks for the quick response , should Reboot, press CTRL+L to boot in Legacy Boot Mode, then press ESC and select your USB from the boot menu... The usb don't get detected on rebooting or after choosing 2TianoCore it boots to chrome os.

MrChromebox, 2022-10-08:
If Tianocore doesn't boot, then the stock firmware is buggy and little I can do. There is no legacy boot mode in newer Chromebooks, it's related by the alternative bootloader menu. SeaBIOS does not work as an alt payload on these devices.

newmonksh, 2022-10-08:
Aha... thank you!!!

closed 2022-10-08 by newmonksh

newmonksh, 2022-10-09:
to add I was able to disable the hardware WP ,update the firmware via option 2 and make it work.

MrChromebox, 2022-10-09:
> to add I was able to disable the hardware WP ,update the firmware via option 2 and make it work.

depends on what you mean by "make it work." You'll gain UEFI booting, but lose the ability to run ChromeOS. You won't be able to run Windows, and Linux will have several functional deficiencies (like built-in audio)