MrChromebox/scripts #229: Suspend on Amber Lake Y

Sonic-Y3k, 2022-10-13:
Hi there,

I'm currently trying to run ArchLinux on a Google Pixel Slate (Nocturne). Most of the things work out of the box, as expected. Sadly suspend and hibernate do not seem to work at all and I'm out of ideas, hoping that one of you guys can point me into the right direction.

I tried debugging suspending on my own for a little and can confirm that this step works:

```bash
echo 1 > /sys/power/pm_debug_messages
echo 1 > /sys/power/pm_trace
echo freezer > /sys/power/pm_test
echo freeze > /sys/power/state
```
but running
```
echo devices > /sys/power/pm_test
echo freeze > /sys/power/state
```
results in an instant freeze of the OS without any chances of recovery.

I also tried booting with these kernel command line parameters to get a few more infos:
```bash
cryptdevice=/dev/mmcblk0p3:cryptlvm 
resume=/dev/mapper/ArchLinux-swap 
iomem=relaxed 
nopat 
mem_sleep_default=deep 
console=tty0 
console=ttyS5,115200 
no_console_suspend 
initcall_debug 
ignore_loglevel
```

but dmesg only states:
```bash
[  832.129945] PM: suspend entry (deep)                                                                                                                                                                                                     
[  832.172271] Filesystems sync: 0.038 seconds                                                                                                                                                                                              
[  832.176957] PM: Preparing system for sleep (deep)                                                                                                                                                                                        
[  832.296948] Freezing user space processes ... (elapsed 0.001 seconds) done.                                                                                                                                                              
[  832.306648] OOM killer disabled.                                                                                                                                                                                                         
[  832.310281] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.                                                                                                                                                         
[  832.319972] PM: Suspending system (deep) 
```

Here are some basic infos:
```
os-version              : Arch Linux
kernel-version          : 6.0.1-arch1-1
baseboard-manufacturer  : Google
baseboard-product-name  : Nocturne
baseboard-serial-number : QC...
baseboard-version       : 1.0
bios-release-date       : 08/05/2022
bios-vendor             : coreboot
bios-version            : MrChromebox-4.17.2
chassis-manufacturer    : Google
processor-manufacturer  : GenuineIntel
processor-version       : Intel(R) Core(TM) m3-8100Y CPU @ 1.10GHz
system-manufacturer     : Google
system-product-name     : Nocturne
system-serial-number    : 8A...
system-version          : 1.0
cpucount                : 4
memtotal                : 8035380 kB
```

```bash
❯ cat /sys/power/mem_sleep
s2idle [deep]

[... 219 lines cut ...]
crypto_simd            16384  1 aesni_intel
sdhci                  90112  1 sdhci_pci
cryptd                 24576  4 crypto_simd,ghash_clmulni_intel
mmc_core              233472  4 sdhci,cqhci,mmc_block,sdhci_pci
cros_ec_keyb           20480  0
hid_google_hammer      20480  1
hid_vivaldi_common     16384  1 hid_google_hammer
vivaldi_fmap           16384  2 hid_vivaldi_common,cros_ec_keyb
matrix_keymap          16384  1 cros_ec_keyb
usbhid                 73728  0
xhci_pci               20480  0
xhci_pci_renesas       24576  1 xhci_pci
uhid                   20480  1
8250_dw                24576  0
intel_lpss_pci         28672  4
intel_lpss             16384  1 intel_lpss_pci
idma64                 20480  0
```
As shutting down / rebooting also does not work, I assume that this a result of an APCI error? 

Here is a quite verbose dmesg:
[dmesg.log](https://github.com/MrChromebox/scripts/files/9780270/dmesg.log)

I really hope that you have an idea what else I can try. Thanks in advance,
Sonic-Y3k

MrChromebox, 2022-10-13:
dmesg shows the keyboard causing a kernel panic. It does not show any ACPI issues.

have you tried booting Ubuntu 2204 (eg) to rule out your setup as the issue?

Sonic-Y3k, 2022-10-13:
Oh wow, now I just feel really stupid. I had seen the kernel panic but thought that the kernel NULL pointer dereference just made sense, as the tablet does not have a keyboard. And kept ignoring it for now.

But after your comment I just tried to blacklist `cros_ec_keyb` and `hid_google_hammer`. And what can I say... Rebooting, shutdown and hibernate now work just perfectly.

Thank you so much! :-)

closed 2022-10-13 by Sonic-Y3k