MrChromebox/scripts #231: Error downloading one or more required files; cannot continue

128732, 2022-10-23:
Hi, I keep receiving this error message I tried reinstalling ChromeOS from a recovery usb however that did not fix the issue. Where do I go from here?

``chronos@localhost / $ cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

firmware-util.sh: warning: firmware-util.sh: warning: script from noexec mount; see https://chromium.googlesource.com/chromiumos/docs/+/master/security/noexec_shell_scripts.md

Downloading supporting files...
Error downloading one or more required files; cannot continue``

MrChromebox, 2022-10-23:
on what device / what ChromeOS version?

128732, 2022-10-23:
Version 103.0.5060.132 (Official Build) (64-bit)
Model Code: XE500C13-K04US

davidbishop, 2023-02-21:
Hello

I'm also newly having this issue.

I'm on a Toshiba Chromebook 2 (CB30-B-104) running GalliumOS (Linux toshiba 4.16.18-galliumos #1 SMP PREEMPT Sun Jun 23 04:14:45 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux) with UEFI bios.

Been running/updating just great (thanks MrChrombox) but now getting...

cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0   185    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  1465  100  1465    0     0   1660      0 --:--:-- --:--:-- --:--:--     0
[sudo] password for david: 

Downloading supporting files...

Getting device/system info...

Unable to read current firmware; cannot continue:

/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/flashrom)
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/flashrom)

My /lib/x86_64-linux-gnu/libc.so.6 is a link to /lib/x86_64-linux-gnu/libc-2.27.so

An apt search libc doesn't give newer than 2.27.

Any help appreciated.

MrChromebox, 2023-02-21:
@davidbishop your issue is completely different -- your OS is too old for the version of flashrom used by my script. Simply live boot another OS (any 2022 release should be fine) and update from there. And consider moving on from GalliumOS, as it's been dead for years now

closed 2023-02-21 by MrChromebox

davidbishop, 2023-02-23:
Hello Mr. Chromebox

Thanks for the reply.

Understood. Tried live booting Puppylinux and nixOS but has same,similar results.

Will leave for now as GalliumOS is based on Xubuntu LTS 18.04, which ends support in April this year. So will decide then if an upgrade needed. 

Thanks for all the great work you've done for Chromebook users.

Stay well....

MrChromebox, 2023-02-23:
Ubuntu 22.04 works fine for flashing

davidbishop, 2023-02-27:
Thanks, need a bigger flash drive for that. Xubuntu fails...