MrChromebox/scripts #232: Firware script reports WP enabled even after I took the screw out of Glimmer board

apsillers, 2022-10-24:
I bought a lot of 1st-gen Lenovo 11e Chromebooks, which uses a GLIMMER board. Over a month ago, I successfully used the script to install full UEFI and convert it to a ChromeOS-free Linux box.

Today, I tried to convert another 11e, and it tells me "Fw WP: Enabled" even after I removed the WP screw. I don't think I've done anything differently with this second machine (but it's possible!), but I noticed the release date on the script output (2022-10-19) was between my last machine and this one, so that seems like the likelier change.

The device is in developer mode and has the WP screw removed. I am running the script from a `Ctrl+Alrt+T` + `shell` terminal. Is there some other step to disabling Firmware WP I've forgotten, or has the script's W detection changed for my board in the last month or two?

My script output is:

```
  ChromeOS Device Firmware Utility Script [2022-10-19] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: Lenovo ThinkPad 11e/Yoga Chromebook (GLIMMER)
** Platform: Intel BayTrail
**  Fw Type: Stock ChromeOS
**   Fw Ver: Google_Glimmer.5216.198.188 (05/27/2018)
**    Fw WP: Enabled
*********************************************************
**      1) Install/Update RW_LEGACY Firmware 
** [WP] 2) Install/Update UEFI (Full ROM) Firmware 
** [WP] 3) Set Boot Options (GBB flags) 
** [WP] 4) Set Hardware ID (HWID) 
** [WP] 5) Remove ChromeOS Bitmaps 
** [WP] 6) Restore ChromeOS Bitmaps 
*********************************************************
```

Trying option #2 anyway tells me, `Hardware write-protect enabled, cannot flash Full ROM firmware.`

MrChromebox, 2022-10-24:
nothing has changed with the script's WP detection in probably two years. Your device simply is still grounding the !WP pin, preventing the software WP from being disabled. There may be some metal still around the screw hole which is creating connectivity between the segments.

apsillers, 2022-10-24:
Okay, that's clear enough! I did have a tough time getting this one out, so that wouldn't surprise me. Worst case I can always run an RW_LEGACY boot on this one system. Thanks!

closed 2022-10-24 by apsillers

MrChromebox, 2022-10-24:
using RW_LEGACY on an EOL system is far from ideal; worst case you could just flash the chip directly using a USB programmer and a chip clip

apsillers, 2022-10-25:
Problem solved, thanks for the tip! I opened the case back up and blew air into the screw port like my NES cartridge wasn't working; it is now fully writeable. 😁