MrChromebox/scripts #233: USB-Ports not working after Coreboot Update

Sonic-Y3k, 2022-10-25:
Hi there,

I just updated by Chromebook (NOCTURNE) from MrChromebox-4.17.2 to MrChromebox-4.18. After that the two internal USB-C-Ports did not work at all, therefore booting a OS was not possible.

I just saw this commit [Roll back KBL boards with USB KB](https://github.com/MrChromebox/scripts/commit/24b70909bad16eb2b6696406411566f10ccd8f9e) and I think it might be the same problem. I therefore recommend reverting the firmware update for NOCTURNE devices aswell.

After downgrading to MrChromebox-4.17.2 again, the USB devices started working again.

How I downgraded (basically following this guide [Unbricking](https://wiki.mrchromebox.tech/Unbricking)):
1. Plug in Suzy-Q Cable & Verify that cable is properly connected
```bash
ls /dev/ttyUSB*
```
> This command should return 3 items: ttyUSB0, ttyUSB1, and ttyUSB2

2. Actuall recovery
```bash
docker run -it --rm --privileged debian:bullseye bash
```
> run docker container

```bash
apt update
apt install -y libusb-1.0-0-dev libpci-dev dmidecode git gcc g++ sudo libftdi-dev wget

# build flashrom
git clone https://chromium.googlesource.com/chromiumos/platform/standalone-hdctools
cd standalone-hdctools
export USER=root
./flashrom install

# get cbfstool
cd /root
wget https://mrchromebox.tech/files/util/cbfstool.tar.gz && tar -zxf cbfstool.tar.gz
```
> build dependencies

```bash
/root/.local/flashrom/build/flashrom -p raiden_debug_spi:target=AP -r badflash.rom
./cbfstool badflash.rom extract -n vpd.bin -f vpd.bin

wget https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-nocturne-mrchromebox_20220805.rom
wget https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-nocturne-mrchromebox_20220805.rom.sha1

sha1sum -c coreboot_tiano-nocturne-mrchromebox_20220805.rom.sha1

./cbfstool coreboot_tiano-nocturne-mrchromebox_20220805.rom add -n vpd.bin -f vpd.bin -t raw
```
> prepare image

```bash
/root/.local/flashrom/build/flashrom -p raiden_debug_spi:target=AP -w coreboot_tiano-nocturne-mrchromebox_20220805.rom
```
> flash rom

adyrosebrigg, 2022-10-25:
Not sure if it's the same issue, but this update also broke usb input on my guado. Not sure if usb storage is effected in my case, but during boot I couldn't press esc to enter the coreboot menu, or change the selected os in rEFInd using either of the usb keyboards I have.

Without input, I was able to boot to my default boot of ubuntu. In ubuntu, usb was loaded and worked fine, but every time I rebooted I could not input anything during the boot process.

Since I had access to linux, restoring a backup of the 7/18/22 firmware and clearing NVRAM restored full functionality. Not sure where to go in further diagnosing this, but let me know if I can help. Thank you.

Sonic-Y3k, 2022-10-27:
Indeed, 4.18.1 fixed all the issues. Thank you very much @MrChromebox 

```bash
❯ sudo dmidecode | more
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.
Table at 0x7AA2D000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: coreboot
        Version: MrChromebox-4.18.1
        Release Date: 10/27/2022
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                PC Card (PCMCIA) is supported
                BIOS is upgradeable
                Selectable boot is supported
                ACPI is supported
                Targeted content distribution is supported
        BIOS Revision: 4.18
        Firmware Revision: 0.0
```