MrChromebox/scripts #236: System doesn't boot from SD card (Apollo Lake, RW_LEGACY firmware)

kovan, 2022-11-01:
I sucessfully flashed the RW_LEGACY firmware on the Chromebook with Intel Apollo Lake chipset.
But whey I try to boot from the SD card (Ctrl+L on white screen), the system beeps and does nothing.
The SD card contains an Ubuntu 22.10 image flashed with Rufus.

Thank you for your time.

MrChromebox, 2022-11-01:
Pressing CTRL+L will boot the legacy boot payload (SeaBIOS), from which you need to select the SD Card from the boot menu. Is that what you are doing? If the device beeps when your press CTRL+L, then either there is no valid RW_LEGACY payload, or the dev_boot_legacy/dev_boot_altfw flag is not set. That flag gets cleared on many CR50 chromebooks if you put the device to sleep when having booted the RW_LEGACY payload

kovan, 2022-11-01:
Just before you answered I restored the system with the official ChromeOS restore utility, re-flashed the firmware, and now it boots fine from the SD Card. I'll save your answer though, thanks.
Now Grub appears, I chose the first option, and the Gnome/Lubuntu (tried both, v22.10) logo appears for a while, BUT then the screen goes to terminal mode and is filled with the message:

> stdin: Not a typewriter

and finally it says:

> Unable to find a medium containing a live file system.
> Attempt interactive netboot from a URL?
> yes no (default yes):

I will try with other distros.

kovan, 2022-11-01:
Manjaro doesn't start up well either, it says:

> Starting version 251.4-1-manjaro
> ERROR: '/dev/disk/by-label/MANJARO_GNOME_2137' device did not show up after 30 seconds...
> (etc.)

kovan, 2022-11-01:
Umm if I plug the SD card in an USB SD card reader and boot from there it works well. 
Strange but working!

MrChromebox, 2022-11-01:
there's no much that can be done on the legacy boot side if the internal SD card isn't working, it's just a bootloader after all.

MrChromebox, 2023-02-04:
closing as not a script issue

closed 2023-02-04 by MrChromebox