MrChromebox/scripts #239: Error disabling software write protect range when trying to install UEFI

lordgasmic, 2022-11-10:
This is a similar error to [this issue](https://github.com/MrChromebox/scripts/issues/175), but I am using the scripts found on mrchromebox.tech.

[image]
[image]

I have tried running the script in both a live-cd and full install of GalliumOS.

MrChromebox, 2022-11-10:
please exit the script and provide the output of the following two commands from the terminal:

`sudo flashrom --wp-status`
`sudo flashrom --wp-range`

Yamboy1, 2022-11-10:
I'm also getting the same error. I have flashed about 20-30 of the same type of chromebook and have only seen this error on 2 of them so far.
[image]

[image]

After doing a bit more digging, it looks like the firmware write protect screw had not been removed, but the script had reported that it had been disabled, as you can see from my screenshot.

lordgasmic, 2022-11-10:
I no longer have access to a `chronos` login since I blew away the partitions when I did a full ISO install of GalliumOS.  I was able to find able to find flashrom in /tmp and here are the results:
[image]
[image]

MrChromebox, 2022-11-10:
ok, so the problem is software WP is enabled but my script isn't disabling/able to disable it. I'll see if any recent changes may have caused this

MrChromebox, 2022-11-10:
@lordgasmic this is what I expected. Your hardware WP is still enabled, preventing the script from disabling software WP.

closed 2022-11-10 by lordgasmic

lordgasmic, 2022-11-10:
For prosperity's sake, I was able to remove the firmware screw and install the UEFI firmware with no issues.  I couldn't quickly find an image or instructions on how to remove the screw for my laptop so I added it here hoping to help others.

Acer Chromebook 15 CB3-532 hardware firmware screw:
[image]

MrChromebox, 2022-11-10:
thanks, I was about to upload it to my site and realized I already had one for Banon, just didn't link it on the devices page. It's there now though :)