MrChromebox/scripts #242: Blooglet Restore Stock Firmware Issue - Error: Image size doesn't match

nandotabr, 2022-11-30:
Hi MrChromebox, thanks for the amazing work. I recently tried restoring the stock firmware for the HP Chromebook 14a (Blooglet) and ran into an issue: "Error: Image size doesn't match: stat 169 bytes, wanted 16777216! FAILED".

Indeed I can see that the downloaded file is only 169 bytes in size in /tmp:
-rw-r--r-- 1 root   root        169 Nov 30 22:57 stock-firmware.rom

Steps to reproduce: run script -> 2) Restore Stock Firmware -> Continue=Y -> firmware backup=N -> Confirm model=Y

Here is the full output from the script, thanks again for the help:

_Do you wish to continue? [y/N] y

Do you have a firmware backup file on USB? [y/N] n

That's ok, I'll download a shellball firmware for you.

Confirm system details:

Device: HP Chromebook 14a
Board Name: BLOOGLET

? [y/N] y

Downloading shellball.blooglet.bin

VPD extracted from current firmware

Merging VPD into recovery image firmware

Restoring stock firmware

flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 6.0.8-060008-generic (x86_64)
flashrom was built with unknown PCI library, GCC 7.5.0, big endian
Command line (5 args): /tmp/flashrom -n -w -o /tmp/flashrom.log /tmp/stock-firmware.rom
Acquiring lock (timeout=180 sec)...
Opened file lock "/run/lock/firmware_utility_lock"
Lock acquired.
disable_power_management: Disabling power management.
Calibrating delay loop... OS timer resolution is 1 usecs, 1018M loops per second, 10 myus = 10 us, 100 myus = 98 us, 1000 myus = 994 us, 10000 myus = 9897 us, 4 myus = 4 us, OK.
Initializing internal programmer
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-00000530
Found coreboot table at 0x00000000.
coreboot table found at 0x79b14000.
coreboot header(24) checksum: 9e30 table(1304) checksum: cadf entries: 42
Vendor ID: Google, part ID: Bloog
Using External DMI decoder.
DMI string chassis-type: "Laptop"
Laptop detected via DMI.
DMI string system-manufacturer: "Google"
DMI string system-product-name: "Blooglet"
DMI string system-version: "rev4"
DMI string baseboard-manufacturer: "HP"
DMI string baseboard-product-name: "Blooglet"
DMI string baseboard-version: "rev4"
get_target_bus_from_chipset() returns 0x10.
Found chipset "Intel Geminilake" with PCI ID 8086:31f0. Enabling flash write... Vendor ID: 8086, Device ID: 3196, BAR: 9121d000

BIOS_CNTL = 0x8b: BIOS Lock Enable: enabled, BIOS Write Enable: enabled
SPI BAR is = 0x9121d000
GCS = 0x8b: BIOS Interface Lock-Down: enabled, Boot BIOS Straps: 0x0 (SPI)
SPIBAR = 0x00007fec90636000 + 0x0000
ich_ generation 18
0x04: 0x0010f800 (HSFSC)
WARNING: SPI Configuration Lockdown activated.
0x08: 0x00000000 (FADDR)
0x50: 0x000042c3 (FRAP)
BMWAG 0x00, BMRAG 0x00, BRWA 0x42, BRRA 0xc3
0x54: 0x00000000 (FREG0: Flash Descriptor)
0x00000000-0x00000fff is read-only
0x58: 0x0f7e0001 (FREG1: BIOS)
0x00001000-0x00f7efff is read-write
0x5C: 0x00007fff (FREG2: Management Engine)
Management Engine region is unused.
0x60: 0x00007fff (FREG3: Gigabit Ethernet)
Gigabit Ethernet region is unused.
0x64: 0x00007fff (FREG4: Platform Data)
Platform Data region is unused.
0x68: 0x0fff0f7f (FREG5: Device Expansion)
0x00f7f000-0x00ffffff is locked
0x84: 0x00000000 (PR0, unused)
0x88: 0x00000000 (PR1, unused)
0x8C: 0x00000000 (PR2, unused)
0x90: 0x00000000 (PR3, unused)
0x94: 0x00000000 (PR4, unused)
0x98: 0x00000000 (PR5, unused)
Reading flash descriptors mapped by the chipset via FDOC/FDOD... done.
=== Content Section ===
FLVALSIG 0x0ff0a55a
FLMAP0   0x00040003
FLMAP1   0x17100208
FLMAP2   0x00000000

--- Details ---
NR          (Number of Regions):                     1
FRBA        (Flash Region Base Address):         0x040
NC          (Number of Components):                  1
FCBA        (Flash Component Base Address):      0x030
ISL         (ICH/PCH Strap Length):                 23
FISBA/FPSBA (Flash ICH/PCH Strap Base Address):  0x100
NM          (Number of Masters):                     3
FMBA        (Flash Master Base Address):         0x080
MSL/PSL     (MCH/PROC Strap Length):                 0
FMSBA       (Flash MCH/PROC Strap Base Address): 0x000

=== Component Section ===
FLCOMP   0x52583ef5
FLILL    0xad604221

--- Details ---
Component 1 density:            16 MB
Component 2 is not used.
Read Clock Frequency:           50 MHz
Read ID and Status Clock Freq.: reserved
Write and Erase Clock Freq.:    reserved
Fast Read is supported.
Fast Read Clock Frequency:      reserved
Invalid instruction 0:          0x21
Invalid instruction 1:          0x42
Invalid instruction 2:          0x60
Invalid instruction 3:          0xad

=== Region Section ===
FLREG0   0x00000000
FLREG1   0x0f7e0001

--- Details ---
Region 0 (Descr.) 0x00000000 - 0x00000fff
Region 1 (BIOS  ) 0x00001000 - 0x00f7efff

prettyprint_ich_descriptor_master: cs=18
=== Master Section ===
FLMSTR1  0x00200300
FLMSTR2  0x02002300
FLMSTR3  0x00000000
FLMSTR4  0x00000000
FLMSTR5  0xffffff00

--- Details ---
      Descr. BIOS ME GbE Plat EC
BIOS    r     rw                
ME      r     r                 
GbE                             
Plat                            
EC      rw    rw  rw  rw  rw  rw

OK.
The following protocols are supported: Programmer-specific.
Probing for Generic HWSEQ chip, 0 kB: Chip identified: GD25LQ128C/GD25LQ128D
Found 1 attached SPI flash chip with a density of 16384 kB.
There are 4096 erase blocks with 4096 B each.
Found GigaDevice flash chip "GD25LQ128C/GD25LQ128D" (16384 kB, Programmer-specific) at physical address 0x0.
This chip may contain one-time programmable memory. flashrom cannot read
and may never be able to write it, hence it may not be able to completely
clone the contents of this chip (see man page for details).
No -i argument is specified, set ignore_fmap.
Reading Status register
Block protection is disabled.
Error: Image size doesn't match: stat 169 bytes, wanted 16777216!
FAILED
restore_power_management: Re-enabling power management.
An error occurred restoring the stock firmware. DO NOT REBOOT!

MrChromebox, 2022-12-01:
file is fine on the server:
`-rw-r--r-- 1 1006 1006 16777216 May 30  2022  shellball.blooglet.rom`

so it's failing to download for some reason, don't have a good explanation

nandotabr, 2022-12-27:
Hi, I've been trying with quite a few different live images but end up having the same issue in which the downloaded file is only 169 bytes in size. Any thoughts? Thanks!

devcarbon-com, 2023-02-23:
@nandotabr I had a similar problem. I was able to work around it by downloading the file in guest user chrome and copying that to the tmp dir. (the Downloads folder is under /home/chromos/u-<random-string-here>/Downloads). Then I just modified the firmware-util script and the bin/firmware script to continue from the sha check without redownloading everything.

nandotabr, 2023-02-25:
> @nandotabr I had a similar problem. I was able to work around it by downloading the file in guest user chrome and copying that to the tmp dir. (the Downloads folder is under /home/chromos/u-/Downloads). Then I just modified the firmware-util script and the bin/firmware script to continue from the sha check without redownloading everything.

Thank you @devcarbon-com. Did a similar thing to run the firmware utility from the command line and it worked.

closed 2023-02-25 by nandotabr

livelifeofftheland, 2023-06-15:
@nandotabr Hello kindly I would love to know what command you used to fix this I have the same exact issue you had. Thanks in advance.

nandotabr, 2023-06-15:
Hi @livelifeofftheland really apologise but I cannot remember the command off the top of my head now. I sold the device a few months ago. But I recall finding the downloaded rom somewhere in /tmp and running the flashrom command manually, with the same syntax that we have on the script. Hope that will help. Thanks!

nandotabr, 2023-06-15:
> @nandotabr Hello kindly I would love to know what command you used to fix this I have the same exact issue you had. Thanks in advance.

This is what I ran: /tmp/flashrom -n -w -o /tmp/flashrom.log /path-to/stock-firmware.rom

try to locate the firmware file on your system and change the path and it should work

livelifeofftheland, 2023-06-16:
Your awesome thank you so much for the help and with such quick response time. I was almost sure that id not get a response back. I highly appreciate you taking your time and using your experiences to help others with the same issue, as I feel anyone that has gotten in this boat and is looking for a way out there are many challenges to face. I've been trying for months now to get chromeos back. I can built my own computer and write my own programs but this was a sure challenge for me left me stumped many times. I'll be testing what info you have given and will upload my results to help guide others aswell. Thanks again.     - Cody

livelifeofftheland, 2023-06-16:
ChromeOS Device Firmware Utility Script [2023-06-15] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: HP Chromebook 14a (BLOOGLET)
** Platform: Intel GeminiLake
**  Fw Type: Full ROM / UEFI
**   Fw Ver: MrChromebox-4.20.0 (05/15/2023)
**    Fw WP: Disabled
*********************************************************
** [WP] 1) Install/Update UEFI (Full ROM) Firmware 
** [WP] 2) Restore Stock Firmware 
**      C) Clear UEFI NVRAM 
*********************************************************
Select a menu option or
R to reboot  P to poweroff  Q to quit 
2

Restore Stock Firmware

Standard disclaimer: flashing the firmware has the potential to
brick your device, requiring relatively inexpensive hardware and some
technical knowledge to recover.  You have been warned.

Do you wish to continue? [y/N] y

Do you have a firmware backup file on USB? [y/N] n

That's ok, I'll download a shellball firmware for you.

Confirm system details:

Device: HP Chromebook 14a
Board Name: BLOOGLET

? [y/N] y

Downloading shellball.blooglet.bin

VPD extracted from current firmware

Merging VPD into recovery image firmware

Restoring stock firmware

flashrom dc8caac9-dirty on Linux 5.19.0-41-generic (x86_64)
flashrom was built with unknown PCI library, GCC 11.3.0, little endian
Command line (10 args): /tmp/flashrom -p internal --ifd -i bios -N -w /tmp/stock-firmware.rom -o /tmp/flashrom.log
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing internal programmer
/sys/class/mtd/mtd0 does not exist
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-00000534
Found coreboot table at 0x00000000.
coreboot table found at 0x79b1b000.
coreboot header(24) checksum: 53d0 table(1308) checksum: 153b entries: 43
Vendor ID: Google, part ID: Bloog
Using Internal DMI decoder.
DMI string chassis-type: "Laptop"
Laptop detected via DMI.
DMI string system-manufacturer: "Google"
DMI string system-product-name: "Blooglet"
DMI string system-version: "rev4"
DMI string baseboard-manufacturer: "HP"
DMI string baseboard-product-name: "Blooglet"
DMI string baseboard-version: "rev4"
Found chipset "Intel Gemini Lake" with PCI ID 8086:3197.
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
[... 144 lines cut ...]
There is only one partition containing the whole address space (0x000000 - 0xffffff).
There are 4096 erase blocks with 4096 B each.
Added layout entry 00000000 - 00ffffff named complete flash
Found GigaDevice flash chip "GD25LQ128C/GD25LQ128D/GD25LQ128E" (16384 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Found GigaDevice flash chip "GD25LQ128C/GD25LQ128D/GD25LQ128E" (16384 kB, Programmer-specific).
This chip may contain one-time programmable memory. flashrom cannot read
and may never be able to write it, hence it may not be able to completely
clone the contents of this chip (see man page for details).
Reading Status register
Block protection is disabled.
Reading ich descriptor... Reading 4096 bytes starting at 0x000000.
done.
Assuming chipset 'Gemini Lake'.
Added layout entry 00000000 - 00000fff named fd
Added layout entry 00001000 - 00f7efff named bios
Added layout entry 00f7f000 - 00ffffff named reg5
restore_power_management: Re-enabling power management.
Using region: "bios".
Error: Image size (169 B) doesn't match the expected size (16777216 B)!
FAILED
Restoring PCI config space for 00:0d:2 reg 0xdc
restore_power_management: Re-enabling power management.
An error occurred restoring the stock firmware. DO NOT REBOOT!

Press [Enter] to return to the main menu.

livelifeofftheland, 2023-06-16:
user@laptop:~$ /tmp/flashrom -n -w -o /tmp/flashrom.log /path-to/stock-firmware.rom
flashrom dc8caac9-dirty on Linux 5.19.0-41-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Error: Extra parameter found.
Please run "flashrom --help" for usage info.
user@laptop:~$ 

This is what its giving me. Im stumped. To mention I do not have a backup of firmware or rom atleast not that i have saved I never planned on going back so never saved the rom or old firmware.

MrChromebox, 2023-06-16:
@livelifeofftheland well your parameter list is malformed, the example provided above which you are copying has the parameters in the wrong order. it should be exactly what flashrom is executing:

`/tmp/flashrom -p internal --ifd -i bios -N -w /tmp/stock-firmware.rom -o /tmp/flashrom.log`

that doesn't help if `/tmp/stock-firmware.rom` is the wrong size though, which is the error you are getting

livelifeofftheland, 2023-06-16:
> @livelifeofftheland well your parameter list is malformed, the example provided above which you are copying has the parameters in the wrong order. it should be exactly what flashrom is executing: [...]

Upon Running as you have suggested I am getting.

Error: opening log file /tmp/flashrom.log failed permission denied

MrChromebox, 2023-06-16:
because you have to run as sudo. but again, check the size of /tmp/stock-firmware.rom -- it's it's not 16MiB, you're going to get the same error flashing manually as via the script

livelifeofftheland, 2023-06-16:
> because you have to run as sudo. but again, check the size of /tmp/stock-firmware.rom -- it's it's not 16MiB, you're going to get the same error flashing manually as via the script

stock-firmware.rom is a size of 169 bytes in the /tmp folder

livelifeofftheland, 2023-06-16:
user@laptop:~$ sudo /tmp/flashrom -p internal --ifd -i bios -N -w /tmp/stock-firmware.rom -o /tmp/flashrom.log
[sudo] password for user: 
flashrom dc8caac9-dirty on Linux 5.19.0-41-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0x79b1b000.
Found chipset "Intel Gemini Lake".
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG5: Device Expansion region (0x00f7f000-0x00ffffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
OK.
Found GigaDevice flash chip "GD25LQ128C/GD25LQ128D/GD25LQ128E" (16384 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading ich descriptor... done.
Using region: "bios".
Error: Image size (169 B) doesn't match the expected size (16777216 B)!
FAILED

MrChromebox, 2023-06-16:
> > because you have to run as sudo. but again, check the size of /tmp/stock-firmware.rom -- it's it's not 16MiB, you're going to get the same error flashing manually as via the script
> 
> stock-firmware.rom is a size of 169 bytes in the /tmp folder

well, that's not going to work now is it? 

I just fixed something on the server side, please retry restoring via the script

livelifeofftheland, 2023-06-16:
> well, that's not going to work now is it?
>
> I just fixed something on the server side, please retry restoring via the script

I'm sorry just to clerify, which script are you recommending me try now?

livelifeofftheland, 2023-06-16:
user@laptop:~$ sudo /tmp/flashrom -p internal --ifd -i bios -N -w /tmp/stock-firmware.rom -o /tmp/flashrom.log
[sudo] password for user: 
sudo: /tmp/flashrom: command not found

MrChromebox, 2023-06-16:
> I'm sorry just to clerify, which script are you recommending me try now?

there's only one script, I'm not sure why there is confusion. https://mrchromebox.tech/#fwscript

livelifeofftheland, 2023-06-16:
> > I'm sorry just to clerify, which script are you recommending me try now?
> 
> there's only one script, I'm not sure why there is confusion. https://mrchromebox.tech/#fwscript

Your a God it worked thank you soooo much!

MrChromebox, 2025-11-16:
> how does a software or chromeos go missing or get damaged i feel like my computer was attacked causing the image go missing

your question has nothing to do with the issue at hand here, or anything to do with my firmware. It's best handled via a ChromeOS support forum or the ChromeOS subreddit