MrChromebox/scripts #243: Segmentation fault: Acer Chromebook 712 (C871) (Kindred)

tkna91, 2022-12-09:
I am unable to update with the following error.

```
./functions.sh: line 334:  3305 Segmentation fault  (core dumped) ${flashromcmd} -r /tmp/bios.bin > /dev/null 2>&1
```

[image]

Environment :
- Acer Chromebook 712 (C871) (Kindred)
- FW: MrChromebox-4.14 07/25/2021
- Battery removed.
- USB Install Image: Ubuntu Server 22.10

Please let me know if you need more information I will get it.

MrChromebox, 2022-12-09:
flashrom crashed, hard to say way. try live booting a non-server ISO? And no need for battery disconnect

tkna91, 2022-12-10:
Thanks, but it still doesn't seem to work.

- Ubuntu 22.10 Desktop (64-bit)
- Battery connected

[image]

tkna91, 2022-12-10:
I did the following and succeeded!
- Ubuntu 20.04.5 Desktop (64-bit)
[image]
[image]

However, the Arch Linux that was originally installed disappeared from the boot target.
I think this can be fixed by repairing it.
[image]

tkna91, 2022-12-10:
eMMC Device seems to have disappeared

[image]

tkna91, 2022-12-10:
Sorry. The following resolved the issue.
You can close it if you like.

https://mrchromebox.tech/static/faq.html

> Q: I'm using your UEFI firmware, installed my OS [Linux], and it still boots to the EFI shell - what do I do?
> A: Sounds like your Linux distro doesn't install the EFI bootloader in the default location - no worries, it's an easy fix:
> 
> Type 'exit' to return to the UEFI settings menu, then select Boot Maintenance Manager. From there, select Boot From File, then navigate to and boot from /EFI/[distro name]/grubx64.efi (where [distro name] will be ubuntu, arch, debian, etc). 

[image]

Thanks for your cool project!

closed 2022-12-10 by MrChromebox