MrChromebox/scripts #245: No Display - Acer ChromeBase CA24I2

wamsparky, 2022-12-29:
So i proceed to the flash the coreboot_tiano-karma-mrchromebox_20221129.rom through NeoProgrammer tool, but after completion and attempt to power on the device there is no display. Only led light to show power is present. Can someone guide me if i did something wrong.

I have Acer ChromeBase CA24I2-5T model which i believe should be the KARMA model

MrChromebox, 2022-12-30:
not aware of any issues with KARMA, but probably only been flashed a handful of times thus far. I can give you another build with a different display init method and see if that helps

wamsparky, 2022-12-30:
Yes please do, I would really appreciate it as I had this enterprise locked and tried KARMA flash. I reflashed the backups but nothing. Should power be going to the board while flashing or no, or can that affect the chip.

MrChromebox, 2022-12-30:
if the stock backup isn't working, sounds like an issue with the flashing. All power should be disconnected from the board when flashing

wamsparky, 2022-12-30:
So what suggestions would you recommend, when I first flashed it, it did come up but with the same settings. No changes. So I was wondering if I was connecting to the wrong IC, but I backed up both IC but wasn't flashable due to incorrect buffer amount so I knew it was the next IC, my model is CA24I2-5T I am wondering if it's the same Karma rom

MrChromebox, 2022-12-31:
a KARMA is a KARMA is a KARMA.

If the stock backup you made isn't working, the issue is with your flashing setup. All power should be disconnected when connecting the chip clip.

wamsparky, 2022-12-31:
So meaning the program might me reading the chip but not connected properly ? As I attempted the flash again but now led light is flashing blue

MrChromebox, 2022-12-31:
as in, you are not reading nor writing the chip properly. Your backup is almost certainly not valid. See if this one works for you:

[coreboot-Google_Karma.11343.22.0.zip](https://github.com/MrChromebox/scripts/files/10327990/coreboot-Google_Karma.11343.22.0.zip)

wamsparky, 2022-12-31:
Will test and get back to you but I find it strange as it successfully save the backup. I will use the neoprogrammer to flash

MrChromebox, 2022-12-31:
just because it read and saved data doesn't mean that data matches what is on the chip

wamsparky, 2022-12-31:
Understood let me try this flash.

wamsparky, 2022-12-31:
> just because it read and saved data doesn't mean that data matches what is on the chip

odd that this data was pulled from the chip. I will erase the chip and flash what you gave

wamsparky, 2022-12-31:
I erase the IC flashed with the provided rom and it still flashes a blue light on power led no when i power on

MrChromebox, 2022-12-31:
did you erase the other chip (EC firmware) by chance?

wamsparky, 2022-12-31:
No i didnt but i did a backup of that IC as well but i couldnt flash that IC as image/buffer didnt match

MrChromebox, 2022-12-31:
please zip/upload the backups of the two SOIC-8 chips that you read

wamsparky, 2022-12-31:
they are .bin files

wamsparky, 2022-12-31:
[Chromebase Dump.zip](https://github.com/MrChromebox/scripts/files/10328100/Chromebase.Dump.zip)

Please see here

wamsparky, 2022-12-31:
thanks also for the support

MrChromebox, 2022-12-31:
those look sane. Are you sure you don't have the display cable disconnected?

wamsparky, 2022-12-31:
display cable is connected to the board i removed and reseated a few times. should i remove all connections when flashing the board and CMOS battery ?

MrChromebox, 2022-12-31:
just external power. CMOS shouldn't matter but ok to remove

MrChromebox, 2022-12-31:
here is a karma UEFI test build with alternate display init method

[coreboot_tiano-karma-mrchromebox_20221231.zip](https://github.com/MrChromebox/scripts/files/10328113/coreboot_tiano-karma-mrchromebox_20221231.zip)

wamsparky, 2022-12-31:
i am still at a lost i am wondering what i messed up. i ever tried different RAM chips and still just flashing blue light. do you know where i could source a board.

.......I will test this one as this model Chromebase is 2021 model. was this first flash legacy flash ?

wamsparky, 2022-12-31:
Ch341a Programmer V2.2.0.10 vs NeoProgrammer V2.2.0.10 is there a difference

wamsparky, 2023-01-01:
still no luck i am going to purchase another flashing board i am wondering if this one is faulty

MrChromebox, 2023-01-01:
I have no idea what NeoPogrammer is, is it a software app? It is recommended to use flashrom under Linux

wamsparky, 2023-01-01:
I was going off a video I saw on YouTube. Do you have a link for flashrom ? As it was those programs I used to creat thos backups

MrChromebox, 2023-01-01:
see https://wiki.mrchromebox.tech/Unbricking

wamsparky, 2023-01-01:
> see https://wiki.mrchromebox.tech/Unbricking

The issue is i am using a windows PC, i guess the 2 software I quoted are for windows environment

wamsparky, 2023-01-02:
question should i be using 25SPI BIOS or 24I2C EEPROM

MrChromebox, 2023-01-03:
25SPI

wamsparky, 2023-01-03:
ok just making sure, i am the new one

MrChromebox, 2023-02-04:
not a script issue; closing

closed 2023-02-04 by MrChromebox