MrChromebox/scripts #246: Unable to read firmware even though booted with kernel parameter iomem=relaxed

arktyk, 2022-12-30:
I have an Asus Chromebox CN62 running Arch Linux.  Even though I add the kernel parameter iomem=relaxed, I still get the following error:

```
Unable to read current firmware; cannot continue.

Either add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu).

See https://www.flashrom.org/FAQ for more info.
```

MrChromebox, 2022-12-31:
latest Arch kernel seems to be giving flashrom some issues; live-boot a distro with a 5.1x kernel to update

jmontleon, 2023-01-08:
This is true of Fedora 37 and 6.0.17 as well.

MrChromebox, 2023-01-09:
there's also a libpci compatibility issue at play here. Still trying to figure out the best way to handle this for both old/newer versions

MrChromebox, 2023-02-04:
I've updated the build of flashrom used by my script under Linux, let me know if that fixes the issue for you

JulienDeveaux, 2023-02-06:
Same error on Ubuntu 20.04
I have set the kernel parameter but no luck (tried with kernel 6.1 and 5.4)

JulienDeveaux, 2023-02-06:
I retryed with the ' cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh' and by cloning the repo but same result :(

MrChromebox, 2023-02-06:
cloning the repo does nothing useful, why would you do that?

running the script on a live boot of Ubuntu 2204/2210 works perfectly here on my CN62, nothing else needed other than installing curl.

Arch isn't going to work due to kernel perms and it being a rolling release / using newer libraries than flashrom is compiled against

MrChromebox, 2023-02-06:
can only see what in Ubuntu?

JulienDeveaux, 2023-02-06:
The usb drives

MrChromebox, 2023-02-06:
* boot to UEFI settings screen
* go to Boot Menu
* If USB not listed
  * ESC to main screen
  * remove and reinsert USB
  * wait 2s
  * Go to Boot Menu - should be listed now

JulienDeveaux, 2023-02-06:
Still not working 
I tried two different usb sticks with different distros on them.
Grub is not seeing them neither. Ubuntu with kernel 6 don't see them but with kernel 5.4 I can see it, so I don't understand
My goad is to put the chromebook back to chromeos, so I tried many different approaches to boot on a live stick before reverting to the iomem=relaxed path

MrChromebox, 2023-02-06:
putting an EOL device back on stock ChromeOS is a bad idea. Flex is the best path forward for Haswell/Broadwell Chromeboxes if you want to keep running ChromeOS.

What firmware version is listed on the main UEFI screen?

MrChromebox, 2023-02-06:
> This is a kaby lake pc so what should I do ?

what's the goal? what are you trying to do?

> 4.12 06/04/2020

shouldn't have any USB booting issues

JulienDeveaux, 2023-02-06:
I am on ubuntu on a chromebook (no chromeos or other os) and I want to wipe it all to come back to full chromeos like a factory reset.
My plan was to put back the old bios and restore chromeos

MrChromebox, 2023-02-06:
pushed an update to the script, should be a bit more informative as to why it fails to read the current firmware - give it a try after a fresh boot (to ensure tmp is clear)

JulienDeveaux, 2023-02-07:
It's clearer now what's going on it tell me : 
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/flashrom)
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/flashrom)

JulienDeveaux, 2023-02-07:
UpgradinguUbuntu made it work 🥳
Thanks a lot for this error message it helps a LOT

jmontleon, 2023-02-07:
I was able to update from 4.19 to 4.19.1 without issue on Fedora 37 with kernel 6.1.9.

MrChromebox, 2023-02-07:
sounds like this is resolved now that the script is a bit more verbose when failing

closed 2023-02-07 by MrChromebox

WASH90, 2023-11-16:
bonjour 
J'ai un problemme similaire, sauf que le mien a une version UIEFI 4.20.0.05/15/2023 
et j'aime bien remettre la version de base de chromeOS en effet j' ai installé windows sur le (chrombox) hp 14A 
mais helas !! je n'arrive plus a excuter flashroom a traversle script  << cd; curl -LOk https://mrchromebox.tech/setup-kodi.sh && sudo bash setup-kodi.sh -k >> sur invite de commade de UBUNTU
J'ai juste un message derreur ci-joint 
merçi d'avance....
[image]