MrChromebox/scripts #247: Technically possible to re-flash stock firmware from ChromeOS?

ThinIce, 2023-01-05:
I've got a second or third hand Pixelbook Go (Atlas) that's clearly been modified in the past as it had an invalid HWID.  It currently has ChromeOS installed. I'm unsure what's been done to it up to now.

I've removed the battery to disable WP and fixed the HWID using your script, so far so good (and thanks!), ChromeOS is happy. 

However, when obtaining a second hand machine I would generally secure erase the drive and reflash all firmware / bios using known good files. This is the first time I've played with a Chromebook.

I see that the scripts are configured such that the full restore stock firmware option ([WP] 2) is only available if booted into linux with a uefi firmware installed (I think). Is it technically possible to reflash / reinstall everything with the battery removed as if I was in that position from the chromeos chronos developer terminal? (I've already flashed the rw_legacy) or is this nonsensical? 

I've learned a lot from your site / repo and https://link.springer.com/chapter/10.1007/978-1-4842-0070-4_5 but this last question about flashing eludes me.

Thanks for your time and work

MrChromebox, 2023-01-05:
of course it's possible, my script just tries to prevent you from doing something unnecessary or dangerous. If you overwrite the stock firmware, the GBB flags, HWID, and serial are all erased. If the stock firmware were corrupted or otherwise tampered with, the RW section wouldn't validate/boot, and the device would be stuck in recovery. If you've performed a recovery, the RW sections have already been updated/reflashed with Google-verified bits. So there's really no point in reflashing the stock firmware.

ThinIce, 2023-01-08:
Thanks again for your time

closed 2023-01-08 by ThinIce