MrChromebox/scripts #250: Edgar CB3-431 Can't enable Legacy; Firmware utility script has RW_Legacy option grayed out

Olyred, 2023-01-30:
Successfully enabled Developer mode, ran CL to enable legacy crossystem dev_boot_altfw=1 (chronos insisted on using **altfw** in place of **legacy**).  On white dev splash screen Ctrl-L still gives 2 beeps, then boots to ChromeOS.  Firmware utility script does not offer the RW_Legacy firmware update option, it's grayed out.  Trying to boot Gallium or Lubuntu, stuck!  Is it hopeless on a Braswell 'book?

Olyred, 2023-01-30:
Solved, apparently.  Even though grayed out,  RW_Legacy firmware update  ran and installed it.  Kinds unintuitive. All good though, Lubuntu running in live session and functionality is good...haven't tested camera yet.

MrChromebox, 2023-01-30:
if it ran, it's a bug and I need to fix it. RW_LEGACY is not supported on EOL devices

closed 2023-01-30 by MrChromebox