MrChromebox/scripts #252: No RW_LEGACY support for eol devices

christhefrog, 2023-02-02:
What's the reason behind disabling RW_LEGACY support for eol devices?
I tried flashing full ROM on my ThinkPad 11e (Ultima) and it worked but i wasn't able to install any Linux due to weird storage errors (lots of hardware interrupt timeouts)
I read some forums and saw someone go around the same problem with using legacy boot so that's what I'd like to try

christhefrog, 2023-02-02:
I think the user should be able to flash to RW_LEGACY on their own responsibility, with warnings about what you've mentioned
The script can be easily edited, so if someone does, they aren't informed about the drawbacks at all

MrChromebox, 2023-02-02:
well, you're welcome to fork my script and edit it as you please. With an ever growing number of devices to support, I'm trying to reduce, not increase, the number of options available for each. And you're not the one the users come to when their battery drains and they can't boot in legacy mode anymore, despite any warnings

closed 2023-02-02 by MrChromebox

markdascher, 2023-03-06:
Can we at least count on the RW_LEGACY files being available for download indefinitely? I don't mind bypassing the script, since I usually do that anyway.

I use my Chrome OS partition as a flexible recovery environment. As long as I'm careful, `dev_boot_signed_only` ensures a clean slate each time.

MrChromebox, 2023-03-06:
no, I wouldn't say you can count on that. I don't want people running decade-old, unsupported Google firmware. Or a years out of date version of ChromeOS

markdascher, 2023-03-08:
Fair enough! I've already gotten what I need, so it won't affect me. Just realize that there may be more weirdos like myself coming out of the woodwork, who appreciated the site mainly for its thoroughly helpful documentation. For better or worse, I may never be comfortable eliminating the Chrome OS safety net. I can tinker without worrying about causing permanent harm, rootkits, etc.

I can certainly think about alternatives if that's a bad idea. Maybe it doesn't matter, if no one else does this.