MrChromebox/scripts #253: Unable to read current firmware

KaziRosib, 2023-02-03:
I have been trying to revert to chromeos on my ASUS C523NA Chromebook. But it always says "Unable to read current firmware", no matter which linux operating system i use. It's always the same result. It always shows :-----

Getting device/system info...
Unable to read current firmware; cannot continue.

Either add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu).

See https://www.flashrom.org/FAQ for more info.

MrChromebox, 2023-02-03:
I'm not able to understand what you are seeing based on your description, so how about you take a pic and post it here?

KaziRosib, 2023-02-04:
Sorry to bother you sir. But the previous issue is fixed somehow. But now facing new problem with creating the recovery media for my model's leptop. But other seems fine. I know its nothing of your concern, but if you help me, then i will be so thankful to you. Whenever I try to create the media this error massage pops up after varification

[image]

MrChromebox, 2023-02-04:
I would just try it on another machine? maybe you're out of space and the file got truncated? hard to say

KaziRosib, 2023-02-04:
Tried in 3 different machine 🙂. With different USB sticks. But the result is the same 😅.

MrChromebox, 2023-02-04:
I have no idea, but this has nothing to do with my firmware or scripts, so closing it.

closed 2023-02-04 by MrChromebox