MrChromebox/scripts #254: "Error occurred flashing the Full ROM firmware"

lukeallister, 2023-02-03:
On Acer chromebook 15 C434T I try to flash UEFI rom

Write protection disabled (battery disconnected)
Developer mode > shell

```
Error running cmd: /usr/sbin/flashrom -p internal --noverify-all -w coreboot_tiano-sand-mrchromebox_20230129.rom >/dev/null 2>&1 -o /tmp/flashrom.log

An error occurred flashing the Full ROM firmware. DO NOT REBOOT!
```

What does this mean and what can I do?

Thanks!

MrChromebox, 2023-02-03:
nothing was flashed so there's no harm. exit script and try again, it should be fixed

lukeallister, 2023-02-03:
Thanks for the fast response!

I have attempted this 5 times, from both ctrl+alt+t > shell and also ctrl+alt+F2. Both give this same error and quit at the same spot.

Any ideas for what I can try?

MrChromebox, 2023-02-03:
I'll test on my APL board here and see what's up

MrChromebox, 2023-02-03:
I just tested here on REEF and no issues. You're on latest build of ChromeOS?

lukeallister, 2023-02-03:
That was it! I just got this device and went right to converting it to a linux box--I didn't get the 'update' notification until a second reboot.

Thank you!

closed 2023-02-03 by lukeallister