MrChromebox/scripts #256: Lillypup/MXLinux (Debian Stable): GLIBC_2.33' not found (required by /tmp/flashrom)

jmpxy, 2023-02-08:
Hello. 

I'm having issues trying to update my UEFI ROM on MXLinux. 

A few weeks back, I used the firmware utility script to flash a new ROM and have loaded various Linux distros since then. 

Now I'm on MXLinux (Debian Stable) and every time I try to run the script, I get this error:
```

Unable to read current firmware; cannot continue:

/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/flashrom)
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/flashrom)
```

My GLIBC is on 2.31, and there's no update available to upgrade it to a new version.

Is there any solution to  this, or should I try using a different Linux distro?

MrChromebox, 2023-02-08:
the solution is to live boot a USB with a newer GLIBC. I'm not inclined to have my script try to build flashrom locally in an attempt to support older kernels/libraries

mjw99, 2023-02-11:
Just for the record, I was seeing this on GalliumOS 3.1 and Debian 11.6 .

MrChromebox, 2023-02-11:
unless I have the script build flashrom locally, there's simply no way I can support running on every distro. Just the way it is, given that the packaged version of flashrom doesn't do what I need

closed 2023-02-11 by MrChromebox