MrChromebox/scripts #259: Error when trying to erase before writing

juanmafont, 2023-02-21:
Hello. I have and Acer C910 C7YQ.

A lot of years ago (2018) I update bios with SEABIOS Legacy MrChromebox rel.1-1.11.0-54

I have linux (Manjaro XFCE) running at this chromebook, but I try to update to UEFI. No way.

Notes: 
   - The readonly screw has been removed from motherboard.
   - Try it with different distros (live usb), same problem.
   - Added the kernel parameter iomen=relaxed  (and nopat) 
   - dmidecode installed
 
The command is:

`sudo ./flashrom -p internal --ifd -i bios -N -w coreboot_tiano-auron_yuna-mrchromebox_20230129.rom`

and the output is:

```
flashrom dc8caac9-dirty on Linux 6.1.11-1-MANJARO (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Opened /dev/mtd0 successfully
Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading ich descriptor... done.
Using region: "bios".
disable_power_management: Failed to open /run/lock/power_override/flashrom.lock for writing: No such file or directory
Reading old flash chip contents... done.
Erasing and writing flash chip... linux_mtd_erase: MEMERASE ioctl call returned -1, error: Unknown error 524
 ERASE_FAILED
FAILED!
Uh oh. Erase/write failed.

```

if launch with -VVV the same command:

> flashrom dc8caac9-dirty on Linux 6.1.11-1-MANJARO (x86_64)
> flashrom is free software, get the source code at https://flashrom.org
> 
> flashrom was built with unknown PCI library, GCC 11.3.0, little endian
> Command line (9 args): ./flashrom -VVV -p internal --ifd -i bios -N -w coreboot_tiano-auron_yuna-mrchromebox_20230129.rom
> Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
> Initializing internal programmer
> get_mtd_info: device_name: "intel-spi", is_writeable: 1, numeraseregions: 0, total_size: 8388608, erasesize: 4096
> Opened /dev/mtd0 successfully
> The following protocols are supported: Programmer-specific.
> Probing for Programmer Opaque flash chip, 0 kB: Added layout entry 00000000 - 007fffff named complete flash
> Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
> Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific).
> page_size=1000
> pre-rounding:  start=0x00000000ff800000, len=0x800000, end=0x0000000100000000
> post-rounding: start=0x00000000ff800000, len=0x800000, end=0x0000000100000000
> programmer_map_flash_region: mapping Opaque flash chip from 0x00000000ff800000 to 0x00007f4ea54ce000
> Reading ich descriptor... done.
> Assuming chipset '8 series Lynx Point'.
> Added layout entry 00000000 - 00000fff named fd
> Added layout entry 00200000 - 007fffff named bios
> Added layout entry 00001000 - 001fffff named me
> page_size=1000
> pre-rounding:  start=0x00007f4ea54ce000, len=0x800000, end=0x00007f4ea5cce000
> post-rounding: start=0x00007f4ea54ce000, len=0x800000, end=0x00007f4ea5cce000
> programmer_unmap_flash_region: unmapped 0x00007f4ea54ce000
> restore_power_management: Re-enabling power management.
> Looking for region "bios"... found.
> Using region: "bios".
> Flash image seems to be a legacy BIOS. Disabling coreboot-related checks.
> disable_power_management: Disabling power management.
> disable_power_management: Failed to open /run/lock/power_override/flashrom.lock for writing: No such file or directory
> page_size=1000
> pre-rounding:  start=0x00000000ff800000, len=0x800000, end=0x0000000100000000
> post-rounding: start=0x00000000ff800000, len=0x800000, end=0x0000000100000000
> programmer_map_flash_region: mapping Opaque flash chip from 0x00000000ff800000 to 0x00007f4ea3ccb000
> Reading old flash chip contents... 0x200000-0x7fffff:R done.
> fix_erasers_if_needed: kept all erasers
> fill_sorted_erasers: found 1 valid erasers
> 200000..3ebfff   1000 x 492 eraser 0
> 3f0000..3f0fff   1000 x 1 eraser 0
> 3f4000..3f5fff   1000 x 2 eraser 0
> 3f8000..3f8fff   1000 x 1 eraser 0
> 400000..41cfff   1000 x 29 eraser 0
> 5ff000..5fffff   1000 x 1 eraser 0
> 604000..76efff   1000 x 363 eraser 0
> 79f000..7d6fff   1000 x 56 eraser 0
> 7fd000..7fffff   1000 x 3 eraser 0
> Erasing and writing flash chip... 0x200000-0x200fff: Elinux_mtd_erase: MEMERASE ioctl call returned -1, error: Unknown error 524
>  ERASE_FAILED
> FAILED!

I have read about remove kernel modules 'spi-intel'  (used by module spi-intel-platform) and I try to remove this kernel modules, but appears others errors.

I remove spi-intel-platform:

> flashrom dc8caac9-dirty on Linux 6.1.11-1-MANJARO (x86_64)
> flashrom is free software, get the source code at https://flashrom.org
> 
> Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
> coreboot table found at 0x7ce3c000.
> Found chipset "Intel Broadwell U Base".
> Enabling flash write... Error accessing ICH RCRB, 0x4000 bytes at 0x00000000fed1c000
> /dev/mem mmap failed: Operation not permitted
> FAILED!
> FATAL ERROR!
> Error: Programmer initialization failed.

 I remove both, spi-intel and spi-intel-platform modules:

> flashrom dc8caac9-dirty on Linux 6.1.11-1-MANJARO (x86_64)
> flashrom is free software, get the source code at https://flashrom.org
> 
> Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
> coreboot table found at 0x7ce3c000.
> Found chipset "Intel Broadwell U Base".
> Enabling flash write... Error accessing ICH RCRB, 0x4000 bytes at 0x00000000fed1c000
> /dev/mem mmap failed: Operation not permitted
> FAILED!
> FATAL ERROR!
> Error: Programmer initialization failed.

Try it with 'flashrom' command version downloaded by script and with distro (manjaro) 'flashrom' command version. Same error.

Any idea ?

Thank you very much.

MrChromebox, 2023-02-21:
sounds maybe like something is still grounding the !WP pin, check the screw hole and make sure no metal making contact between the two sides

what does `sudo /tmp/flashrom --wp-status` report

juanmafont, 2023-02-21:
> sounds maybe like something is still grounding the !WP pin, check the screw hole and make sure no metal making contact between the two sides
> 
> what does `sudo /tmp/flashrom --wp-status` report

Hi, MrChromebox, thank you.

Screw recess cleaned with contact cleaner. ->
[Photo here](https://photos.app.goo.gl/qsqbW3JaYj6L4fAi6
)
I executed: 

`sudo /tmp/flashrom --wp-status`

> flashrom dc8caac9-dirty on Linux 6.1.12-1-MANJARO (x86_64)
> flashrom is free software, get the source code at https://flashrom.org
> 
> Please select a programmer with the --programmer parameter.
> To choose the mainboard of this computer use 'internal'. Valid choices are:
> internal, dummy, nic3com, nicrealtek, gfxnvidia, raiden_debug_spi, drkaiser,
> satasii, serprog, buspirate_spi, dediprog, rayer_spi, nicintel, nicintel_spi,
> ogp_spi, satamv, linux_mtd, linux_spi

So try: 

`sudo /tmp/flashrom -p internal --wp-status`

> flashrom dc8caac9-dirty on Linux 6.1.12-1-MANJARO (x86_64)
> flashrom is free software, get the source code at https://flashrom.org
> 
> Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
> Opened /dev/mtd0 successfully
> Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
> mtd_wp_status: ioctl: Operation not supported

Maybe I have select another "programer" like **linux_mtd** or **linux_spi** or whatever ? ( I don't know, I'm a complete novice in these matters.)

MrChromebox, 2023-02-21:
yes, the `-p internal is needed`

your kernel is seemingly preventing flashrom from accessing the flash chip registers. maybe try `-p internal:ich_spi_mode=hwseq` ?

juanmafont, 2023-02-22:
> 
 It seems as if the parameter is not recognised.

> tmp]$ sudo ./flashrom -p internal:ich_spi_mode=hwseq --ifd -i bios -N -w coreboot_tiano-auron_yuna-mrchromebox_20230129.rom
> flashrom dc8caac9-dirty on Linux 6.1.12-1-MANJARO (x86_64)
> flashrom is free software, get the source code at https://flashrom.org
> 
> Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
> Opened /dev/mtd0 successfully
> Unhandled programmer parameters: ich_spi_mode=hwseq
> Aborting.
> Error: Programmer initialization failed.

juanmafont, 2023-02-23:
Hi, when I do:

`cat /proc/mtd
`

> dev:    size   erasesize  name
> mtd0: 00800000 00001000 "intel-spi"
> mtd1: 00800000 00001000 "BIOS"

 Maybe flashrom try to use mtd0 (intel-spi) and must use mtd1(BIOS) ? ( I don't know about this, i'm full newbie)

juanmafont, 2023-02-23:
SOLVED ¡¡¡
It was a config or parameter into the kernel. I boot from live usb (Linux Lite) and all OK, now I have UEFI boot.

Thank you MrChromebox for your script and patient.

closed 2023-02-23 by MrChromebox

stephencox, 2023-03-18:
I have a similar issue on a Dell Chromebook P22T:
`flashrom dc8caac9-dirty on Linux 6.2.6-arch1-1 (x86_64)
flashrom was built with unknown PCI library, GCC 11.3.0, little endian
Command line (10 args): /tmp/flashrom -p internal --ifd -i bios -N -w coreboot_tiano-candy-mrchromebox_20230129.rom -o /tmp/flashrom.log
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing internal programmer
get_mtd_info: device_name: "intel-spi", is_writeable: 1, numeraseregions: 0, total_size: 8388608, erasesize: 4096
Opened /dev/mtd0 successfully
The following protocols are supported: Programmer-specific.
Probing for Programmer Opaque flash chip, 0 kB: Added layout entry 00000000 - 007fffff named complete flash
Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific).
Reading ich descriptor... done.
Assuming chipset 'Baytrail'.
Added layout entry 00000000 - 00000fff named fd
Added layout entry 00200000 - 007fffff named bios
Added layout entry 00001000 - 001fffff named me
restore_power_management: Re-enabling power management.
Using region: "bios".
Flash image seems to be a legacy BIOS. Disabling coreboot-related checks.
disable_power_management: Disabling power management.
disable_power_management: Failed to open /run/lock/power_override/flashrom.lock for writing: No such file or directory
Reading old flash chip contents... 0x200000-0x7fffff:R done.
fix_erasers_if_needed: kept all erasers
fill_sorted_erasers: found 1 valid erasers
604000..72dfff   1000 x 298 eraser 0
7fe000..7fffff   1000 x 2 eraser 0
Erasing and writing flash chip... 0x604000-0x604fff: Elinux_mtd_erase: MEMERASE ioctl call returned -1, error: Unknown error 524
 ERASE_FAILED
FAILED!
Uh oh. Erase/write failed. 
Your flash chip is in an unknown state.
`

stephencox, 2023-04-08:
For anyone else having a similar issue, I had to do these steps.

Step 1, add `iomem=relaxed` to your kernel command line and reboot.

Step 2, check the output of `lsmod`. If `spi-intel-platform` is present, unload with `modprobe -r spi-intel-platform`.

Then run the update script again. @MrChromebox it might be worth adding that to the script?