MrChromebox/scripts #263: Enrollment Error Chrome OS Flex

dhiner, 2023-03-15:
After running your scripts to convert a chromebox HP Chromebox G1 to allow Chrome OS Flex, I'm trying to enterprise enroll the device.  I'm getting the message: "Oops! The system failed to establish the device installation-time attributes lock"  I've run the commands to remove TPM settings and that still doesn't seem to work.  What can I do to resolve this issue? Thank you for your help!

MrChromebox, 2023-05-17:
not a script issue, but this is resolved in the just-release MrChromebox-4.20.0 firmware

closed 2023-05-17 by MrChromebox

dhiner, 2023-05-17:
Hello,

Thank you for the update.  When will this be available?  I ran this script
today cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash
firmware-util.sh   and it is showing the 4.19.1 firmware instead of 4.20?
Or how do I get it?

Thanks again
_______________________________

Dave Hiner ***@***.***)
Network Administrator / Computer Services
Willoughby-Eastlake City Schools
35353 Curtis Blvd.
Eastlake, Ohio 44095
Phone: (440) 942-3615
Fax:    (440) 946-4671

MrChromebox, 2023-05-17:
what do you mean "it is showing"?

the script menu shows your current firmware version, as well as if an update is available (with the date), but does not show the new firmware version (because it has no way of knowing that)