MrChromebox/scripts #268: DELL 7410 BUG

geochensx, 2023-04-28:
Hello! My computer is a Dell 7410 laptop. I have tested the latest version you released. The battery works well in the win10 state, but the touchpad and touch screen do not respond after installing the driver. Some people say that the touch version of 20230123 can work normally, I wonder whether this problem will be fixed in the later version?
What puzzles me is that basically all of the hardware works when installing fydeos.
My English is not good, I hope you can understand.

closed 2023-05-06 by geochensx

MrChromebox, 2023-05-06:
Any issues with the touchpad/touchscreen on DRALLION in the 4.19.1 release will be fixed in the upcoming 4.20 release

geochensx, 2023-05-07:
Thank you for your response and thank you for your efforts on DELL 7410