MrChromebox/scripts #269: Unable to read current firmware

Saturn345, 2023-05-02:
I'm trying to sell a HP Chromebook 11 G3 (kip) but when I run the Firmware script to remove SeaBIOS I get this error. (Unable to read current firmware)
I don't really know why this is happening @MrChromebox.

OS: Ubuntu 18.04 LTS (Live)
Version currently on: 07/25/2021

MrChromebox, 2023-05-03:
> I'm trying to sell a HP Chromebook 11 G3 (kip)

putting back on EOL ChromeOS isn't the best idea, and isn't supported by the script FYI.

> to remove SeaBIOS

you mean the UEFI firmware?

> I don't really know why this is happening @MrChromebox.
> OS: Ubuntu 18.04 LTS (Live)

if you'd google the error it would tell you - flashrom, used by my script, requires linux libraries that are newer than what Ubuntu 18.xx includes. You can live boot a Ubuntu 22.xx USB to run the script from there.

Saturn345, 2023-05-21:
ok thx

closed 2023-10-23 by MrChromebox