MrChromebox/scripts #270: Touch and Sound not working in legacy mode HP x360 14a (BLOOGUARD) chromebook

apandada1, 2023-05-03:
I have a HP x360 14a (BLOOGUARD) Chromebook with Gemini Lake (Celeron 4020) processor.

I enabled developer mode, removed the battery to remove write protection, and updated `RW_LEGACY firmware` using the `firmware-util.sh` script.

I booted Manjaro KDE (manjaro-kde-22.1.0-230421-linux61.iso) and Ubuntu 23.04.

In both these operating systems, the touchscreen is not working, and audio is showing a dummy output (also the internal mic is not working).

Is this known behavior? If I flash the UEFI Full ROM firmware, will it support the touch screen and the audio out of the box?

apandada1, 2023-05-03:
Update: Sound works in legacy mode (tested in Manjaro) after using the script in https://github.com/eupnea-linux/audio-scripts. Mic works after increasing the volume in alsamixer.

Only the touch screen is not working right now.

MrChromebox, 2023-05-03:
touchscreen will almost certainly work OOTB with the UEFI Full ROM; RW_LEGACY is just a bootloader, so you're booting Google's stock firmware, which almost certainly includes ACPI code for multiple touchscreen models, which confuses the linux kernel.

audio is just a function of OS support/drivers/Alsa etc

MrChromebox, 2023-05-17:
closing as this is not a script issue

closed 2023-05-17 by MrChromebox

MrChromebox, 2025-09-02:
@MonsterSe7en this isn't a script issue. please see https://docs.chrultrabook.com for OS help / audio fix script

archisman-panigrahi, 2025-09-02:
The repo has moved to https://github.com/WeirdTreeThing/chromebook-linux-audio

https://askubuntu.com/questions/1486278/how-to-install-chromebook-audio-drivers-in-ubuntu