MrChromebox/scripts #275: Blooglet Restore Stock Firmware Issue - Error: Image size doesn't match #242

mandarsdroid, 2023-05-17:
The firmware utility does not work in my case and fails as the error mentioned in the Above issue reported previously. 

I have a HP Chromebook 14-na0003tu
There was similar issue present previously. 

I managed to get the firmware image using the 
https://github.com/coreboot/coreboot/blob/master/util/chromeos/crosfirmware.sh

This script calls a recovery.conf located at 
https://dl.google.com/dl/edgedl/chromeos/recovery/recovery.conf
This file has 3 set of BLOOG* configuration all of them referring to following recovery image 

"https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_15359.58.0_octopus_recovery_stable-channel_mp-v32.bin.zip"

I flashed my chromebook with this firmware so it is detected as "HP Chromebook x360 12b"
**My chromebook works but the camera does not work and mic does not work.** 

There must be another recovery image for the said chromebook but I cannot find the link. Can someone help me find correct recovery firmware for HP Chromebook 14-na000tu

MrChromebox, 2023-05-17:
> There must be another recovery image for the said chromebook 

absolutely not. Every Geminilake chromebook is built off the "octopus" baseboard and uses the same octopus recovery image. That recovery image contains the firmware for every Geminilake Chromebook. 

the crosfirmware.sh script selects the firmware image for your device based on the board name provided. If you ran `./crosfirmware.sh blooglet` then you got the correct image. What you don't have is the correct HWID, because recovery image firmwares do not have those. That's one of the many reasons to not lose your firmware backup.

> I flashed my chromebook with this firmware so it is detected as "HP Chromebook x360 12b"

no, it's not. ChromeOS has no idea about make/model names. Only board names. Your HWID is wrong.

closed 2023-05-17 by MrChromebox

mandarsdroid, 2023-05-17:
Is there a way I could get the firmware or correct HWID for HP 14a-na0003tu, 

No matter what I could not get camera and mic to work.

or is there any other work around?  I just need the camera and Mic to work.