MrChromebox/scripts #277: Acer C710-2834 (PARROT) - multiple flash chips detected when updating firmware

oberon4071, 2023-05-18:
On my Acer C710-2834 (PARROT), I attempted to run the `firmware-util.sh` script to upgrade the firmware from coreboot 4.19 to coreboot 4.20. However, the script was not able to proceed because `flashrom` detected multiple flash chips when attempting to perform the firmware updated.

More info:

Machine/board: Acer C710-2834 (PARROT)
CPU: Intel Celeron 1007U (Ivy Bridge)
Host OS used for running firmware script: Kubuntu 23.04 live ISO
Firmware script command: `cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh`

Here is the relevant `flashrom` output demonstrating this issue:

```
Unable to read current firmware; cannot continue:

flashrom dc8caac9-dirty on Linux 6.2.0-20-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xacf4c000.
Found chipset "Intel NM70".
Enabling flash write... SPI Configuration is locked down.
The Flash Descriptor Override Strap-Pin is set. Restrictions implied by
the Master Section of the flash descriptor are NOT in effect. Please note
that Protected Range (PR) restrictions still apply.
OK.
Found Macronix flash chip "MX25L6405" (8192 kB, SPI) mapped at physical address 0x00000000ff800000.
Found Macronix flash chip "MX25L6405D" (8192 kB, SPI) mapped at physical address 0x00000000ff800000.
Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) mapped at physical address 0x00000000ff800000.
Found Macronix flash chip "MX25L6436E/MX25L6445E/MX25L6465E/MX25L6473E/MX25L6473F" (8192 kB, SPI) mapped at physical address 0x00000000ff800000.
Multiple flash chip definitions match the detected chip(s): "MX25L6405", "MX25L6405D", "MX25L6406E/MX25L6408E", "MX25L6436E/MX25L6445E/MX25L6465E/MX25L6473E/MX25L6473F"
Please specify which chip definition to use with the -c <chipname> option.
```

MrChromebox, 2023-05-19:
known issue with sandybridge/ivybridge platform devices and the updated flashrom build the script is currently using.

to workaround, run from home dir:
`sudo rm /tmp/flashrom`
`curl -LO https://mrchromebox.tech/files/util/flashrom.tar.gz`
`tar zxf flashrom.tar.gz`
`mv flashrom /tmp`
`sudo bash firmware-util.sh`

that should resolve the issue

oberon4071, 2023-05-25:
Thank you for the workaround! I needed to also add ```chmod +x /tmp/flashrom``` after the ```mv``` command, as the file's permissions weren't set to execute.

With these steps, I was able to proceed to update the firmware, with a segfault in the ```flashrom``` step as shown in the output below:

```
Installing Full ROM firmware (may take up to 90s)

flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 6.2.0-20-generic (x86_64)
./firmware.sh: line 166:  5354 Segmentation fault      (core dumped) ${flashromcmd} -L -o /dev/null > /dev/null 2>&1
Full ROM firmware successfully installed/updated.
```

After shutting down and restarting the machine, the firmware still successfully updated to version 4.20 despite the above segfault. I have experienced no new issues thus far with the updated firmware.

harshula, 2023-10-19:
Hi, The workaround that I used was to explicitly provide the chipname as an argument to `flashrom`. e.g.
```
--- a/functions.sh
+++ b/functions.sh
@@ -409,7 +409,7 @@ if [[ "$isChromeOS" = true || "$isChromiumOS" = true ]]; then
        gbbutilitycmd=$(which gbb_utility)
 else
        #set cmds
-       flashromcmd=/tmp/flashrom
+       flashromcmd=/tmp/flashrom -c ${chipname}
        cbfstoolcmd=/tmp/cbfstool
        gbbutilitycmd=/tmp/gbb_utility
 fi

```

MrChromebox, 2023-10-19:
closing as this issue should be solved by the latest version of flashrom used by the script

closed 2023-10-19 by MrChromebox