MrChromebox/scripts #280: APELE Device?

SKlein-1428, 2023-05-31:
I just got an Asus CX1101CMA which is confusing me with what seems to be old technology on a new board?

It's a GeminiLike platform, so the built in RW_LEGACY refuses to see anything other than the ChromeOS partition, but because APELE is not supported I can't update the RW_LEGACY or install the UEFI firwmare.

What's confusing me even more is that there's no jumper for WP. It gets disabled just by removing the battery which, I may be misunderstanding, but implies an older board?

When I searched for APELE in the supported devices list it only came up with APEL. I do not know nearly enough to just assume they're the same and change the hardware ID and try to force things for APEL to work, so, I'm not. Gonna do that.

All I really want to do is install a proper Linux distro so if it's possible to enable support for APELE, or whatever this device is, even just to get a working RW_LEGACY I would be greatly appreciative.

Any ideas?

SKlein-1428, 2023-05-31:
Someone else has decided they want it, I'll go looking for a fully supported second hand device.

MrChromebox, 2023-05-31:
first I've heard of APELE, but it's in the recovery list so I'll add support for it

MrChromebox, 2023-05-31:
added