MrChromebox/scripts #281: Acer c710-2845 error in update

antonvier, 2023-06-02:
Hello, after trying to install the start-os of start9 in this device that ran for a long time with the dietpi, there was an error and the machine only starts via usb. I tried to fix this by updating the firmware, but got the following error:

...
Unable to read current firmware; cannot continue:

flashrom dc8caac9-dirty on Linux 6.2.8-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xacf6a000.
Found chipset "Intel NM70".
Enabling flash write... Error accessing ICH RCRB, 0x4000 bytes at 0x00000000fed1c000
/dev/mem mmap failed: Operation not permitted
FAILED!
FATAL ERROR!
Error: Programmer initialization failed.

MrChromebox, 2023-06-02:
duplicate of #277 -- see the workaround listed there

closed 2023-06-02 by MrChromebox

antonvier, 2023-06-02:
don't work, i tried.

MrChromebox, 2023-06-02:
> /dev/mem mmap failed: Operation not permitted

means your kernel is locked down / not allowing flashrom access. Boot a permissive kernel like Ubuntu and update from there using the WA in the linked issue

antonvier, 2023-06-02:
Thanks @MrChromebox , works fine now.

Congratulations on your brilliant project!