MrChromebox/scripts #283: HP Elite Dragonfly can not flash firmware

leenchan, 2023-06-15:
flashrom 1.4.0-devel on Linux 5.10.180-22627-g679a94b8d670 (x86_64)
flashrom was built with LLVM Clang 17.0.0 (/mnt/host/source/src/third_party/llvm-project 98f5a340975bc00197c57e39eb4ca26e2da0e8a2), little endian
Command line (10 args): /usr/sbin/flashrom -p internal --ifd -i bios -N -w coreboot_edk2-redrix-mrchromebox_20230515.rom -o /tmp/flashrom.log
Acquiring lock (timeout=180 sec)...
Opened file lock "/run/lock/firmware_utility_lock"
Lock acquired.
Initializing internal programmer
/sys/class/mtd/mtd0 does not exist
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-000006e8
Found coreboot table at 0x00000000.
coreboot table found at 0x7688b000.
coreboot header(24) checksum: da96 table(1744) checksum: 8cb5 entries: 54
Vendor ID: Google, part ID: Redrix
Using Internal DMI decoder.
DMI string chassis-type: "Laptop"
Laptop detected via DMI.
DMI string system-manufacturer: "Google"
DMI string system-product-name: "Redrix"
DMI string system-version: "rev3"
DMI string baseboard-manufacturer: "Google"
DMI string baseboard-product-name: "Redrix"
DMI string baseboard-version: "rev3"
Found chipset "Intel Alder Lake-P" with PCI ID 8086:51a4.
Enabling flash write... BIOS_SPI_BC = 0x1900008b: BIOS Interface Lock-Down: enabled, Boot BIOS Straps: 0x0 (SPI)
Top Swap: not enabled
SPI Read Configuration: prefetching enabled, caching enabled, 
BIOS_CNTL = 0x8b: BIOS Lock Enable: enabled, BIOS Write Enable: enabled
Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
SPIBAR = 0x00007cc8be721000 (phys = 0x807c8000)
0x04: 0xf000 (HSFS)
HSFS: FDONE=0, FCERR=0, AEL=0, BERASE=0, SCIP=0, FDOPSS=1, FDV=1, FLOCKDN=1
SPI Configuration is locked down.
Reading OPCODES... done
        OP        Type      Pre-OP
op[0]: 0x01, write w/o addr, none
op[1]: 0x02, write w/  addr, none
op[2]: 0x03, read  w/  addr, none
op[3]: 0x05, read  w/o addr, none
op[4]: 0x20, write w/  addr, none
op[5]: 0x9f, read  w/o addr, none
op[6]: 0xd8, write w/  addr, none
op[7]: 0x0b, read  w/  addr, none
Pre-OP 0: 0x06, Pre-OP 1: 0x50
0x06: 0x0010 (HSFC)
HSFC: FGO=0, FCYCLE=0, FDBC=0, SME=0
0x08: 0x00000001 (FADDR)
0x0c: 0x00001f00 (DLOCK)
DLOCK: BMWAG_LOCKDN=0, BMRAG_LOCKDN=0, SBMWAG_LOCKDN=0, SBMRAG_LOCKDN=0,
       PR0_LOCKDN=1, PR1_LOCKDN=1, PR2_LOCKDN=1, PR3_LOCKDN=1, PR4_LOCKDN=1,
       SSEQ_LOCKDN=0
0x50: 0x000042c7 (FRAP)
BMWAG 0x00, BMRAG 0x00, BRWA 0x42, BRRA 0xc7
0x54: 0x00000000 FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
0x58: 0x1fff0500 FREG1: BIOS region (0x00500000-0x01ffffff) is read-write.
0x5C: 0x04ff0001 FREG2: Management Engine region (0x00001000-0x004fffff) is read-only.
0x60: 0x00007fff FREG3: Gigabit Ethernet region is unused.
0x64: 0x00007fff FREG4: Platform Data region is unused.
0x68: 0x00007fff FREG5: Device Expansion region is unused.
0x6C: 0x00007fff FREG6: BIOS2 region is unused.
0x70: 0x00007fff FREG7: unknown region is unused.
0x74: 0x00007fff FREG8: EC/BMC region is unused.
0x78: 0x00007fff FREG9: unknown region is unused.
0x7C: 0x00007fff FREG10: unknown region is unused.
0x80: 0x00007fff FREG11: unknown region is unused.
0xE0: 0x00007fff FREG12: unknown region is unused.
0xE4: 0x00007fff FREG13: unknown region is unused.
0xE8: 0x00007fff FREG14: unknown region is unused.
[... 274 lines cut ...]
Reading Status register
HSFC: FGO=1, FCYCLE=0, FDBC=0, SME=0
ich_hwseq_read_status: only supports STATUS1
wp_read_register: read from register 2 not is supported by programmer, writeprotect operations will assume it contains 0x00.
Reading Status register
HSFC: FGO=1, FCYCLE=0, FDBC=0, SME=0
write_wp_bits: wp_verify reg:1 value:0x0
ich_hwseq_read_status: only supports STATUS1
wp_read_register: read from register 2 not is supported by programmer, writeprotect operations will assume it contains 0x00.
write_wp_bits: wp_verify reg:2 value:0x0
Reading Status register
HSFC: FGO=1, FCYCLE=0, FDBC=0, SME=0
ich_hwseq_read_status: only supports STATUS1
wp_read_register: read from register 2 not is supported by programmer, writeprotect operations will assume it contains 0x00.
Reading Status register
HSFC: FGO=1, FCYCLE=0, FDBC=0, SME=0
write_wp_bits: wp_verify reg:1 value:0x0
ich_hwseq_read_status: only supports STATUS1
wp_read_register: read from register 2 not is supported by programmer, writeprotect operations will assume it contains 0x00.
write_wp_bits: wp_verify reg:2 value:0x0
restore_power_management: Re-enabling power management.
Using region: "bios".
Error: Image size (8388608 B) doesn't match the expected size (33554432 B)!
FAILED
Restoring PCI config space for 00:1f:5 reg 0xdc

MrChromebox, 2023-06-15:
REDRIX image files corrected/updated, please exit script and try again

leenchan, 2023-06-16:
> REDRIX image files corrected/updated, please exit script and try again

Downloading Full ROM firmware
(coreboot_edk2-redrix-mrchromebox_20230615.rom)

Firmware download checksum fail; download corrupted, cannot flash.

i have try to download coreboot_edk2-redrix-mrchromebox_20230615.rom.sha1, but 404.  missed updating sha1?

MrChromebox, 2023-06-16:
sorry the checksum file didn't get uploaded, try again now plz

leenchan, 2023-06-16:
> sorry the checksum file didn't get uploaded, try again now plz

thank so much. bro. now successfully to flash. test full ROM now.

closed 2023-06-17 by leenchan