MrChromebox/scripts #290: ASUS CN60 UEFIFw: Kodi Freezez after 12Sec

12devm, 2023-07-09:
Hi, Asus CN60 ,did the Bios update to UEFI- Thank you for great work. Unfortunately,  video freezes exactly after 12 sec. I'm not sure what info to provide.Kodi-Lbreelec- did all the video upscaling changes etc; reinstalled many times. Please help.I'm not a Linux expert but can do some command line work. Any hints or guidance is much appreciated.Thank you. Dv
[putty_1.log](https://github.com/MrChromebox/scripts/files/11995150/putty_1.log)

MrChromebox, 2023-07-09:
there's nothing in the log to show a problem, need to look at the kodi log file I guess. I've not used LE myself in many years, so your best bet for support is going to be the LE or Kodi forums. Likely a kernel bug or similar.

12devm, 2023-07-09:
> there's nothing in the log to show a problem, need to look at the kodi log file I guess. I've not used LE myself in many years, so your best bet for support is going to be the LE or Kodi forums. Likely a kernel bug or similar.

Thank you. Will do that.

MrChromebox, 2023-10-14:
Closing as not a script issue

closed 2023-10-14 by MrChromebox