MrChromebox/scripts #299: run the script in root user?

xcpky, 2023-08-17:
on the firmware utility website, it's documented:
> Also, you must execute these commands **as a normal/non-root user**. Running them as root will break things. DO NOT RUN 'SUDO SU' BEFORE RUNNING THE SCRIPT CMD BELOW.

A recent chromeOS dev update seems to disable sudo usage for chronos user.  see https://www.reddit.com/r/chromeos/comments/15tnj8i/i_cannot_use_sudo_commands_in_crosh/  https://www.reddit.com/r/Crouton/comments/15aiy80/sudo_cant_enter_chroot_after_update_26_july_2023/

But in VT-2 I can login as root user to use privileged command. What will break if I run the script in root user? And can I modify the script to make it work?

MrChromebox, 2023-08-17:
I'll need to look into this. For now, you can try replacing the `cd;` in the script with `cd /tmp` and `sudo bash firmware-util.sh` with `bash firmware-util.sh`

MrChromebox, 2023-08-18:
according to the folks at Google I talked to, this change only affects running things via a crosh shell (CTRL+ALT+t).

You'll still be able to use sudo from a VT2 terminal (CTRL+ALT+F2, login as chronos)

xcpky, 2023-08-18:
> according to the folks at Google I talked to, this change only affects running things via a crosh shell (CTRL+ALT+t).
> 
> 
> 
> You'll still be able to use sudo from a VT2 terminal (CTRL+ALT+F2, login as chronos)

Unfortunately, this is not true. I can't use sudo in VT2 either, and I get the same error message as in crosh.

MrChromebox, 2023-08-18:
that's not how it supposed to be set up, I'll file a bug so hopefully it doesn't make it to stable channel that way

MrChromebox, 2023-08-18:
in the meantime, don't use dev channel

Dae-Hang-Sook, 2023-10-05:
> according to the folks at Google I talked to, this change only affects running things via a crosh shell (CTRL+ALT+t).
> 
> You'll still be able to use sudo from a VT2 terminal (CTRL+ALT+F2, login as chronos)

This just does nothing. Either 'emerge-' or 'cros' is not recognized by the system. Clearly, this has not been built as it should have.

ari-shi, 2023-10-06:
> Unfortunately, this is not true. I can't use sudo in VT2 either, and I get the same error message as in crosh.

Hello, from my side i was able to use the scripte on the VT2 terminal by using the commande without  the first cd, to commande worked with curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh on an Asus Chromebox 4

Dae-Hang-Sook, 2023-10-06:
> Hello, from my side i was able to use the scripte on the VT2 terminal by using the commande without  the first cd, to commande worked with curl -LO mrchromebox.tech/firmware-util.sh && sudo bash [...]

Probably should've mentioned my device is arm64

MrChromebox, 2023-10-14:
closing as the solution to running the script in R117+ is to use VT2

closed 2023-10-14 by MrChromebox

Simbaclaws, 2023-10-17:
Sorry for posting a comment here, but does anyone know what the default password for chronos would be? No password doesn't work.... And it's asking me to login to the shell via VT2

Legomountain14, 2024-08-22:
I'm having an issue. No password for root and chronos are both not working. What should i do?

MrChromebox, 2024-08-22:
> I'm having an issue. No password for root and chronos are both not working. What should i do?

power wash and not enable things you don't understand next time (ie, do NOT enable ChromeOS Debugging options)