MrChromebox/scripts #304: Remove ChromeOS Bitmaps on KabyLake (EVE)?

jlariviere1, 2023-08-26:
I love this firmware script!  I do wish the Remove ChromeOS Bitmap was available on other devices, such as my 2017 PixelBook (EVE).  
Thanks!

MrChromebox, 2023-08-26:
removing the bitmaps on newer chromebooks is a lot more complicated and not something I plan to add support for unfortunately

aarondill, 2023-10-02:
Nevermind. Having *finally* scraped the bottom of the internet, I came across these resources (below), which indicate that this would be a *difficult* task, rather than just complicated. Sadly, this seems to be a difficult topic to google effectively (I only finally found results with `"vbgfx.bin" extract`), which lead me to spend a lot of time just trying to find other instances of people trying the same.

https://github.com/hexdump0815/linux-mainline-on-arm-chromebooks/blob/main/readme.md
https://github.com/hexdump0815/imagebuilder/blob/61a5297fde1431bf28ff390aeabe6b4eab52f223/todo.txt#L9-L14

MrChromebox, 2023-10-02:
I've never had the time or inclination to look into exactly what's needed to unpack/modify/repack the vbgfx.bin file

closed 2023-10-13 by MrChromebox