MrChromebox/scripts #306: Firmware Utility Script automatically selects the wrong (but almost right) firmware board image (BLOOGLET)

PrimeyThePenguin, 2023-09-01:
Hey there! I have an HP Chromebook 14a-na0031wm (BLOOGLET) and have had the UEFI Full ROM firmware for a few months now and I'm totally loving it! There has been a big drawback for my specific case, however, and it took me this long to find any leads. If I could get expert input on the matter, this might be the place to go.

Anyway, I just realized that the Firmware Utility Script picked the firmware for board BLOOG, rather than BLOOGLET. Even subsequent firmware updates are doing the same thing. The hardware is similar enough, sure, but I wonder if my sound issues are caused by this.
Firstly, sound when booted normally (booted into Linux Mint Xfce 21.2 through the integrated eMMC) works, but no matter what you set your volume to, it's on full blast, all the way (unless it's muted, of course.) The headphone jack doesn't work at all and isn't detected, however, and sound while booted from USB (Live environment, Tails, Batocera, etc.) doesn't work at all, either.

Is this a unique issue, and can I try fixing it by manually selecting a BLOOGLET image somehow?

MrChromebox, 2023-09-01:
BLOOG and BLOOGLET are the exact same device, only difference is the name.

Functional audio under Linux requires the use of the script linked in my FAQ. Has nothing to do with the firmware, it's an OS/driver/config issue

PrimeyThePenguin, 2023-09-01:
Thanks for the tip! I'll give the audio script a whirl when I get home and follow up

closed 2023-09-01 by PrimeyThePenguin