MrChromebox/scripts #307: Legacy BIOS Not Appearing in altfw Menu

Sonic-Flash, 2023-09-05:
The only boot loaders in altfw Menu are the following:
1. U-Boot 
2. Tianocore

But when Tianocore is selected, only replies with 2 beeps.

I Have checked and my Chromebook is supported for Mr Chromebox's firmware script, but when I check the code, there seems like there is no code for editing the altfw list.I may be mistaken, since Tianocore is suppose to be added to my Chromebook's Firmware, Right?

My Hardware ID: "**Treeya**"

**Please fix this.**

Sonic-Flash, 2023-09-05:
Also, Just in case, I have an AMD Based Chip, The x86-64 Architecture. 
Information from Chrome OS Diagnostics: https://github.com/MrChromebox/scripts/assets/107735896/9272f2b0-90d0-407e-aacc-306e41f74d51

Sonic-Flash, 2023-09-05:
Firmware Script Running: https://github.com/MrChromebox/scripts/assets/107735896/d1bec1c5-4204-4214-a699-d3bf5f7f7bdd

Note: From The Video, You Can See I Still Have Firmware Write Protect On.

closed 2023-09-05 by MrChromebox

Sonic-Flash, 2023-09-10:
Am I able to disable hardware write protect and set up full UEFI to fix this?

MrChromebox, 2023-09-10:
You can flash the UEFI firmware, but AMD Stoneyridge devices don't run Linux well, or Windows at all. You should leave the device on ChromeOS.

Sonic-Flash, 2023-09-12:
Thanks, I only wanted Kali Linux.I'll disable write-protect and get Full-UEFI. I don't want Windows, I Actually Hate it!