MrChromebox/scripts #309: Asus Chromebook flip 15 cm5500

tfc31, 2023-09-21:
bonjour j'ai besoin de votre aide svp j'ai le Asus j'ai voulu installer  Windows 10 j'arrive pas  merci d'avance sachant que jarrive pas a faire recovery merci 
[image]
[image]

MrChromebox, 2023-09-21:
Windows support is completely handled by coolstar via coolstar.org and the chrultrabook discord server.

closed 2023-09-21 by MrChromebox

tfc31, 2023-09-21:
Merci de tas réponse si possible me dire comment faire comme vous le voyer sur la photos je peut pas installer update uefi  full rom firmware

MrChromebox, 2023-09-21:
https://coolstar.org/chromebook/windows-install.html

tfc31, 2023-09-23:
bonjour j'ai suivi les étape voila se que j'ai comme le devise error et même j'ai rentrer dans recovery utility pour faire pareille je peut même pas restaurer  a l'etat  d'usine  et j'ai regarder sur YouTube d'autre 
[image]
[image]
marque comme hp il faut enlever un vis de sécurité sur la carte mère merci d'avance 
[image]