MrChromebox/scripts #310: VPD gets lost...

EMLommers, 2023-09-26:
When running script from ChromeOS and verifying VPD it is blank. Stock -> UEFI full rom
Using the script again to return to stock, only the product name is a residue.  
E.g. Babymega THI-SSH-OUL-DBE becomes Babymega 123-ABC-456-DEF

MrChromebox, 2023-09-27:
the HWID string is not in the VPD. The script has never persisted that, because you should be restoring from your own backup.

don't know what you mean by "EC gets noisy" - electrical noise? fan noise? something else?

vbt is unchanged from stock other than removing the fixed (1024x768) resolution on boot

EMLommers, 2023-09-27:
I can remember in the past hwid was unchanged.
OK was just what I noticed. Just a FYI.

Electrical noise. Processing noise. Like coil-whine, but different. But maybe only noticeable for HSPs. 🤔
Chromebooks are famous for additional unwanted noise.

closed 2023-09-27 by EMLommers

MrChromebox, 2023-09-27:
#296 is completely unrelated, that user didn't even use my firmware/script at all!

> Of course I used my backup. The VDP (RO_VPD) is only 0x00 or 0xFF. This cannot be true?

please attach or email me the backup file you restored via USB so I can try and see what's going on