MrChromebox/scripts #311: Full ROM firmware for coreboot_edk2-galith-mrchromebox_20230721.rom is failed with checksum issue

Arulvadivel, 2023-09-28:
My asus cx1500cka is failed with checksum failed at stage:
"Downloading Full ROM firmware"
coreboot_edk2-galith-mrchromebox_20230721.rom"

Could you please confirm whether the above .rom file exists in your repo or not?

Because , I see the other post below which got fixed by some correction in the mrchrome
server side ? https://www.reddit.com/r/chrultrabook/comments/gxp25q/firmware_download_checksum_fail_coreboot/

Arulvadivel, 2023-09-29:
Thanks a lot quick check and really appreciate your help , MrChromeBox.

In my previous try, it was failed and does not downloaded rom.sha1 file.
Manually, I downloaded both rom and rom.sha1 in /tmp directory. Again, started the firmware-util.sh script, it works fine now.

**Just sharing one learning(which may usual for anyone) from ASUS CX1500CKA to unlock the firmware write-protect is that:**
'sudo flashrom --wp-status' and 'sudo flashrom --wp-disable'  is required to disable the WP after disconnecting the battery.

Only disconnecting the battery will not unlock/disable the WP.

The following installation steps help me to install windows 11 on Chromebook ASUS CX1500CKA.
https://www.youtube.com/watch?v=wwE7UlWbJHE

closed 2023-09-29 by Arulvadivel

MrChromebox, 2023-09-29:
> 'sudo flashrom --wp-status' and 'sudo flashrom --wp-disable' is required to disable the WP after disconnecting the battery.
>
> Only disconnecting the battery will not unlock/disable the WP.

that is flat out incorrect. the script literally executes those commands as needed. it is *never* necessary to disable software WP manually before flashing