MrChromebox/scripts #317: Can't restore stock firmware

Sk44rt, 2023-10-12:
HP Chromebook 11A G6 EE (Barla)
AMD A4-9120C

A few weeks ago i've installed coreboot firmware and linux on my chromebook

but i got an issue that there's no sound (coreboot doesn't initialize soundcard)
and i one more issue that my system lags
to avoid that i need to wait until CPU temperature reaches 30°+ then reboot
so i want to restore stock firmware

when i tried to do that i've got an error mmap failed few times
fixed with kernel parameter iomem=relaxed
script says that stock firmware successfully installed
but when i reboot i see coreboot, what can i did wrong?

MrChromebox, 2023-10-12:
> but i got an issue that there's no sound (coreboot doesn't initialize soundcard)

this isn't a firmware issue, it's a Linux/driver one. There are fixes. Same with the GPU. You can't expect everything to work out of the box

> UPDATE: i've fixed this issue with reverting firmware to version MRCHROMEBOX-4.20
4.21 makes impossible to flash stock ROM

this doesn't make any sense, why would 4.21 "allow" you to flash 4.20 but not go back to stock?

Heroeshigh, 2023-10-12:
Another confusing thing I can't to go back to stock too. The script said success, but reboot to coreboot. And I saw this issue I updated the rom from 4.18 to 4.21. Successed. 
That means it allow me to update full rom but not go back to stock. And I got all abot this with my pixelbook 2017.

Sk44rt, 2023-10-12:
i'm sure it's firmware problem cuz i tried to build kernel with needed drivers, but no changes

Sk44rt, 2023-10-12:
> this doesn't make any sense, why would 4.21 "allow" you to flash 4.20 but not go back to stock?

i'm not a low-level programmer to say how it works, idk

Sk44rt, 2023-10-12:
> Another confusing thing I can't go back to stock too. The script said success, but reboot to coreboot. And I saw this issue I updated the rom from 4.18 to 4.21. Successed. That means it allow me to update full rom but not go back to stock. And I got all abot this with my pixelbook 2017.

you can revert
you need to clone scripts repo, then git checkout 66d8709
copy firmware.sh, functions.sh and sources.sh to /tmp, and remove
```
echo -e "\nDownloading supporting files..."
rm -rf firmware.sh >/dev/null 2>&1
rm -rf functions.sh >/dev/null 2>&1
rm -rf sources.sh >/dev/null 2>&1
$CURL -sLO ${script_url}firmware.sh
rc0=$?
$CURL -sLO ${script_url}functions.sh
rc1=$?
$CURL -sLO ${script_url}sources.sh
rc2=$?
if [[ $rc0 -ne 0 || $rc1 -ne 0 || $rc2 -ne 0 ]]; then
	echo -e "Error downloading one or more required files; cannot continue"
	exit 1
fi
```
to prevent script from updating *.sh files, and now you can run firmware-util.sh, revert firmware to 4.20 (now you need to reboot) and restore stock firmware

Sk44rt, 2023-10-12:
> this isn't a firmware issue, it's a Linux/driver one. There are fixes. Same with the GPU. You can't expect everything to work out of the box
i tried a lot of fixes, with GPT, from google/github
nothing works

MrChromebox, 2023-10-12:
> i tried a lot of fixes, with GPT, from google/github nothing works

that's not useful info. I'm literally sitting next to someone right now with a Stoneyridge Chromebook which has working audio and GPU acceleration under Linux.

Sk44rt, 2023-10-12:
> > i tried a lot of fixes, with GPT, from google/github nothing works
> 
> that's not useful info. I'm literally sitting next to someone right now with a Stoneyridge Chromebook which has working audio and GPU acceleration under Linux.

so how it was fixed ?

MrChromebox, 2023-10-12:
I just tested flashing to stock and restoring on several Chromebooks and cannot reproduce this issue

Heroeshigh, 2023-10-12:
I will try the solution Sk44rt provided. And if it not work, I will record it and mail to MrChromebox. Maybe something goes wrong with the chip choosing I guess, because I tried to unbrick another pixelbook with flashing full or stock firmware, flashrom ask me to choose between two ship that meet the flashing.

MrChromebox, 2023-10-12:
> because I tried to unbrick another pixelbook with flashing full or stock firmware, flashrom ask me to choose between two ship that meet the flashing.

I recently changed the script to use the distro-supplied flashrom binary if it's available and can read the chip. It's possible that there is an issue with writing vs reading, but I don't see how the script would be reporting a successful flash if that's not the case.

there is no way that what version of the firmware the device is running is having any effect, so you're wasting your time flashing an older version (and that's only going to work until I get around to archiving the older versions)

Heroeshigh, 2023-10-13:
Well, switch to LinuxMint installed not the live USB. It turns out to use the distro-supplied flashrom then I get this:
`flashrom v1.2 on Linux 5.15.0-76-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0x7aaa3000.
Found chipset "Intel Kaby Lake Y w/ iHDCP2.2 Prem.".
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... Warning: Setting Bios Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG2: Management Engine region (0x00001000-0x001fffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
OK.
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading ich descriptor... done.
Using region: "bios".
Reading flash... done.
An error occurred restoring the stock firmware. DO NOT REBOOT!
`

Heroeshigh, 2023-10-13:
And full rom installation is normal working.

Heroeshigh, 2023-10-13:
Finally I restored stock firmware sucessed with the former scripts which allowed to restore with the stock firmware extracted from the recovery image. So maybe there is nothing wrong with flashrom and the UEFI firmware, but the script and the shellball firmware.

Sk44rt, 2023-10-13:
> > i tried a lot of fixes, with GPT, from google/github nothing works
> 
> that's not useful info. I'm literally sitting next to someone right now with a Stoneyridge Chromebook which has working audio and GPU acceleration under Linux.

How was fixed audio and GPU acceleration ?
and on what distro used ?

MrChromebox, 2023-10-13:
> How was fixed audio and GPU acceleration ? and what distro used ?

I don't believe it's documented anywhere currently. Ask for help in the Chrultrabook discord

closed 2023-10-13 by MrChromebox

Sk44rt, 2023-10-14:
> I don't believe it's documented anywhere currently. Ask for help in the Chrultrabook discord

u just said u sitting right to one who have all those workin on stoneyridge

MrChromebox, 2023-10-14:
> > I don't believe it's documented anywhere currently. Ask for help in the Chrultrabook discord
> 
> u just said u sitting right to one who have all those workin on stoneyridge

that person is on the Chrultrabook discord. I'm not playing middleman here