MrChromebox/scripts #318: do the scripts work with linux or only chromeos?

hmm5, 2023-10-12:
Hi, I have installed xubuntu 23.10, and wrote over the chromeos partition on the emmc after running the script to allow legacy boot.  I forgot to set the boot default while I still had chromeos to automatically load the legacy boot rather than the default chromeos boot on startup.

I tried running the script while under xubuntu, and it doesn't seem to work correctly, stating that there's issues with the software write protect.  I have disabled the battery as I did before, to allow the legacy boot the first time, so that shouldn't be an issue.

MrChromebox, 2023-10-12:
what's the issue exactly?

hmm5, 2023-10-13:
I have already installed xubuntu and have already overwritten chromeos on the emmc.  But I want it to default boot using legacy boot.  

I tried to run the script again using xubuntu, and when I try to set the gbb flags for boot options, it says the there's an error disabling software write-protect; unable to set gbb flags.

The script says that my Fw: WP is disabled and the [WP] in the menu are green.  

I tried to flashrom --wp-status, and instead it says "please select a programmer with --programmer parameter"?  Maybe  xubuntu 23.10 flashrom is too new?

MrChromebox, 2023-10-13:
> I tried to flashrom --wp-status, and instead it says "please select a programmer with --programmer parameter"? Maybe xubuntu 23.10 flashrom is too new?

no, you simply need to provide a programmer with the `-p` option. You should use `-p internal`

I think I need to add a verbose option to the script so that I can better debug these sorts of flashrom problems

hmm5, 2023-10-13:
Hi, I tried with -p internal.  Does this look correct to you?

```
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Cannot open file stream for /dev/mtd0
coreboot table found at 0x79b28000.
Found chipset "Intel Gemini Lake".
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG1: BIOS region (0x00001000-0x00f7efff) is read-write.
FREG5: Device Expansion region (0x00f7f000-0x00ffffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
OK.
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) on internal.
Failed to get WP status: WP operations are not implemented for this chip

```
------
Using the verbose mode -V setting

```
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
flashrom was built with GCC 12.3.0, little endian
Command line (4 args): flashrom -V -p internal --wp-status
Initializing internal programmer
get_mtd_info: device_name: "BIOS", is_writeable: 0, numeraseregions: 0, total_size: 16777216, erasesize: 4096
Cannot open file stream for /dev/mtd0
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-000005d4
Found coreboot table at 0x00000000.
coreboot table found at 0x79b28000.
coreboot header(24) checksum: 6f78 table(1468) checksum: f8f0 entries: 45
Vendor ID: Google, part ID: Phaser
Using Internal DMI decoder.
DMI string chassis-type: "Laptop"
Laptop detected via DMI.
DMI string system-manufacturer: "Google"
DMI string system-product-name: "Phaser"
DMI string system-version: "rev5"
DMI string baseboard-manufacturer: "Google"
DMI string baseboard-product-name: "Phaser"
DMI string baseboard-version: "rev5"
Found chipset "Intel Gemini Lake" with PCI ID 8086:3197.
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... BIOS_SPI_BC = 0x9: BIOS Interface Lock-Down: disabled, Boot BIOS Straps: 0x0 (SPI)
Top Swap: not enabled
SPI Read Configuration: prefetching enabled, caching enabled, 
BIOS_CNTL = 0x09: BIOS Lock Enable: disabled, BIOS Write Enable: enabled
SPIBAR = 0x00007fb88c3a3000 (phys = 0xc1121000)
0x04: 0x6001 (HSFS)
HSFS: FDONE=1, FCERR=0, AEL=0, BERASE=0, SCIP=0, FDOPSS=1, FDV=1, FLOCKDN=0
Programming OPCODES... done
0x06: 0x3f0a (HSFC)
HSFC: FGO=0, FCYCLE=1, FDBC=63, SME=0
0x0c: 0x00000000 (DLOCK)
DLOCK: BMWAG_LOCKDN=0, BMRAG_LOCKDN=0, SBMWAG_LOCKDN=0, SBMRAG_LOCKDN=0,
       PR0_LOCKDN=0, PR1_LOCKDN=0, PR2_LOCKDN=0, PR3_LOCKDN=0, PR4_LOCKDN=0,
       SSEQ_LOCKDN=0
0x50: 0x000042c3 (FRAP)
BMWAG 0x00, BMRAG 0x00, BRWA 0x42, BRRA 0xc3
0x54: 0x00000000 FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
0x58: 0x0f7e0001 FREG1: BIOS region (0x00001000-0x00f7efff) is read-write.
0x68: 0x0fff0f7f FREG5: Device Expansion region (0x00f7f000-0x00ffffff) is locked.
0x7C: 0x7fff7fff FREG10: unknown region (0x07fff000-0x07ffffff) has unknown permissions.
0x80: 0x7fff7fff FREG11: unknown region (0x07fff000-0x07ffffff) has unknown permissions.
0xE0: 0x7fff7fff FREG12: unknown region (0x07fff000-0x07ffffff) has unknown permissions.
0xE4: 0x7fff7fff FREG13: unknown region (0x07fff000-0x07ffffff) has unknown permissions.
0xE8: 0x7fff7fff FREG14: unknown region (0x07fff000-0x07ffffff) has unknown permissions.
0xEC: 0x7fff7fff FREG15: unknown region (0x07fff000-0x07ffffff) has unknown permissions.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
0xa0: 0xc0 (SSFS)
SSFS: SCIP=0, FDONE=0, FCERR=0, AEL=0
0xa1: 0xfe0000 (SSFC)
SSFC: SCGO=0, ACS=0, SPOP=0, COP=0, DBC=0, SME=0, SCF=6
0xa4: 0x5006     (PREOP)
0xa6: 0x463b     (OPTYPE)
0xa8: 0x05200302 (OPMENU)
0xac: 0xc79f0190 (OPMENU+4)
0xc4: 0xb1d82024 (LVSCC)
LVSCC: BES=0x0, WG=1, WSR=0, WEWS=0, EO=0x20, VCL=1
0xc8: 0x00002000 (UVSCC)
UVSCC: BES=0x0, WG=0, WSR=0, WEWS=0, EO=0x20
Enabling hardware sequencing by default for Apollo/Gemini/Jasper/Elkhart/Meteor Lake.
OK.
The following protocols are supported: Programmer-specific.
Probing for Programmer Opaque flash chip, 0 kB: Hardware sequencing reports 1 attached SPI flash chip with a density of 16384 kB.
Added layout entry 00000000 - 00ffffff named complete flash
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) on internal.
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific).
Failed to get WP status: WP operations are not implemented for this chip
Restoring MMIO space at 0x7fb88c3a30ac
Restoring MMIO space at 0x7fb88c3a30a8
Restoring MMIO space at 0x7fb88c3a30a6
Restoring MMIO space at 0x7fb88c3a30a4

```

MrChromebox, 2023-10-13:
looks fine outside of flashrom not supporting WP operations - probably too old a version. Let me add a check for that in my script

MrChromebox, 2023-10-13:
added a check in my script for using the built-in flashrom, give the script a try again and see if it correctly downloads a flashrom binary to use / script now works for setting GBB flags

hmm5, 2023-10-13:
Hi, now the script is saying ""error reading firmware (non-stock?); unable to set boot options".

I tried to see in /tmp, using the script's downloaded flashrom, rather than ubuntu's version to disable the wp, using --wp-disable, and it does say "Disabled hardware protection."   It also says "Found GigaDevice flash chip "GD25LQ128C/GD25LQ128D/GD25LQ128E" (16384 kB, Programmer-specific) on internal," rather than "Opaque flash chip".  So it does look like it's seeing the flash chip correctly with this flashrom version.  

BTW.  I also can't "Install/Update RW_LEGACY Firmware", if that matters, it throws up an error.

MrChromebox, 2023-10-13:
attach the /tmp/flashrom.log file please (do not paste contents)

hmm5, 2023-10-13:
Here's the flashrom.log
[flashrom.log](https://github.com/MrChromebox/scripts/files/12897490/flashrom.log)

MrChromebox, 2023-10-13:
> Here's the flashrom.log [flashrom.log](https://github.com/MrChromebox/scripts/files/12897490/flashrom.log)

that log shows a successful read, the script should not have reported a failure

MrChromebox, 2023-10-13:
added a fix, should work now hopefully

hmm5, 2023-10-13:
Hmm..  It's still throwing up the same errors.

MrChromebox, 2023-10-13:
try again, sometimes you don't always get the latest version from github right away

hmm5, 2023-10-13:
Unfortunately still the same error keeps showing up.  I tried to deleting the downloaded scripts in /tmp, to see if it made a difference, but still the same error showed up.

MrChromebox, 2023-10-13:
just pushed another update to output the flashrom error

hmm5, 2023-10-13:
Now it says "Error: Extra Parameter found."

MrChromebox, 2023-10-13:
is the script using the distro-provided flashrom or downloaded one? the command line path in the error should show which

hmm5, 2023-10-13:
I just uninstalled the previously distro-provided flashrom to make sure it's not being used.  

Still the same error.  The version being used is flashrom 1.4.0-devel (git:v1.2-1362-g49832e3c73) on Linux 6.5.0-9-generic (x86_64)

MrChromebox, 2023-10-13:
ok that's my downloaded version. I'm not sure why there's a problem but will look into it

MrChromebox, 2023-10-13:
ok, pushed another fix

hmm5, 2023-10-13:
Unfortunately still getting "Error: Extra Parameter found." and afterwards "error reading firmware (non-stock?); unable to set boot options".

hmm5, 2023-10-13:
[[ "$isChromeOS" = "true" ]] && FMAP="" || FMAP="--fmap-file /tmp/bios/bin"

should it be? 

[[ "$isChromeOS" = "true" ]] && FMAP="" || FMAP="--fmap-file /tmp/bios.bin"

----------

Also trying to manually execute "./flashrom -r -i GBB:/tmp/gbb.temp --fmap-file /tmp/bios.bin" still gets an extra parameter error.  Do you need to use the "-l" option with --fmap-file?

hmm5, 2023-10-13:
-r needs an output file, this also requires -p internal.

I was able to manually execute "./flashrom -r /tmp/tempbios -i GBB:/tmp/gbb.temp --fmap-file /tmp/bios.bin -p internal", and finally write a tempbios file.

GamePlayer-8, 2023-10-14:
Also should I clear the UEFI NVRAM?

MrChromebox, 2023-10-14:
> [[ "$isChromeOS" = "true" ]] && FMAP="" || FMAP="--fmap-file /tmp/bios/bin"
> 
> should it be?
> 
> [[ "$isChromeOS" = "true" ]] && FMAP="" || FMAP="--fmap-file /tmp/bios.bin"

 yes, will fix

> I was able to manually execute "./flashrom -r /tmp/tempbios -i GBB:/tmp/gbb.temp --fmap-file /tmp/bios.bin -p internal", and finally write a tempbios file.

that is not a valid command string, you can't use both `-r <output file>` and `-i <region>:<output file>`

hmm5, 2023-10-14:
Not sure what you did, but the script is working now.  I was able to set the gbb flags.

Thanks!

MrChromebox, 2023-10-14:
I switched back to using CrOS flashrom and adjusted the script accordingly :)

closed 2023-10-14 by MrChromebox