MrChromebox/scripts #319: `device` variable returns invalid name while flashing on Linux instead of Chrome OS.

GamePlayer-8, 2023-10-13:
Good morning,

I've got a problem with the `firmware-util.sh` script provided by the MrChromebox.
Whenever I'm trying to flash a full rom onto my Chromebook, I'm getting:
`The script does not currently have a firmware file for your device (GRUNT); cannot continue.`
where `GRUNT` is the name returned from `device` variable, initialized from `functions.sh`:
```shell
device=$(dmidecode -s system-product-name | tr '[:upper:]' '[:lower:]' | sed 's/ /_/g' | awk 'NR==1{print $1}')
```

The Chromebook what I have is `HP Chomebook 11A G6 EE` while I'm trying to flash the rom on Linux (`Artix Linux`) instead of Chrome OS.

I think, I can bypass the issue through command:
```shell
curl -LO https://mrchromebox.tech/files/firmware/full_rom/coreboot_edk2-barla-mrchromebox_20230930.rom && flashrom -p internal -w coreboot_edk2-barla-mrchromebox_20230930.rom
```

Although I'm not sure if the motherboard would boot without setting up serial number / `hwid` etc.

MrChromebox, 2023-10-13:
the script has special handling to read the HWID from the GBB region when a device is booted to Linux while running the stock firmware - that should set the board name correctly. Post a pic of the main script menu please

GamePlayer-8, 2023-10-13:
Of course, here's the screenshot:
[image]

GamePlayer-8, 2023-10-13:
In addition: Your script is using `dmidecode` & `flashrom` provided by Arch Linux repositories.
It may not contain features like `flashrom --wp-status`. Is that might be an issue?

MrChromebox, 2023-10-13:
I just pushed a fix for setting the device variable. I also pushed a change to check for --wp-status support before determining to use the distro-provided flashrom build. See if those resolve the issue

GamePlayer-8, 2023-10-13:
An error occured during the flash.
Here's the [log](https://gist.github.com/GamePlayer-8/21620eaf4518dc56197bb553a66d80e7).

Utility says, to do not reboot.

GamePlayer-8, 2023-10-13:
```log
This coreboot image (Google:Careena) does not appear to
be correct for the detected mainboard (Google:Grunt).
Aborting. You can override this with -p internal:boardmismatch=force.
```

Should I be worried about it?

MrChromebox, 2023-10-13:
nothing to worry about, it simply didn't try to flash due to the board name mismatch. No worries on rebooting. I'll figure out how to work around this

MrChromebox, 2023-10-13:
pushed a fix, give it a try plz

GamePlayer-8, 2023-10-14:
It says, it flashed properly - [log](https://gist.github.com/GamePlayer-8/ee96bb704fbc034d9680bf0019868676).

Although what if after reboot it wouldn't boot?

GamePlayer-8, 2023-10-14:
Also should I clear the UEFI NVRAM?

GamePlayer-8, 2023-10-14:
After reboot, chromebook has successfully booted (NVRAM was cleaned automatically).

Closing issue.