MrChromebox/scripts #322: not flashing

Bobsisf, 2023-10-15:
```
ChromeOS Device Firmware Utility Script [2023-10-14] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: Lenovo ThinkPad 11e/Yoga Chromebook (G3) (ULTIMA)
** Platform: Intel Braswell
**  Fw Type: Full ROM / UEFI
**   Fw Ver: MrChromebox-4.21.0 (09/30/2023)
**    Fw WP: Disabled
```

then after selecting USB got this:
```
Extracting firmware from recovery USB

Stock firmware successfully extracted from ChromeOS recovery image

VPD extracted from current firmware

Merging VPD into recovery image firmware

Restoring stock firmware

Stock firmware successfully restored.

After rebooting, you will need to restore ChromeOS using a
	ChromeOS recovery USB, then re-run this script to reset the
	Firmware Boot Flags (GBB Flags) to factory default.

```

Restarted and nothing shanged.

MrChromebox, 2023-10-15:
> When i try to run script again without reboot, have more options, restore flags, change HWID, etc, but every of this chois finished with errors.. Option Clear NVRAM finished with error- something with region .

YOU HAVE TO REBOOT FIRST!!!

The stock firmware is not active until you reboot. You can't  use any of those functions until you reboot. I need to update the script to disable all functions after flashing or restoring apparently

Bobsisf, 2023-10-15:
Running script and flash with no errors, but when reboot bios started with rabbit again. Starting linix live again and there is the same option like first time- update uefi, flash stock and clear NVram.. I mean its seem starting from beginning.. Nothing flashed. 

Thats why i used to run script again without rebooting, then in second run are new command options.

Bobsisf, 2023-10-15:
Should i play with PK keys in bios? My Chromebook is too old, have reached EOL

MrChromebox, 2023-10-15:
if your chromebook is EOL, why on earth are you trying to go back to stock firmware? The script explicitly tells you this is not a good idea

Bobsisf, 2023-10-15:
Is that mean no way to go stock? Is inly 16GB EMMC, and cant install nothing useful as chrome, except Gallium ..

MrChromebox, 2023-10-15:
I just fixed the issue with the wrong firmware image being flashed.

I don't recommend going back to stock because no more updates. Did you try using Brunch?

Bobsisf, 2023-10-15:
Didn't tryed with Brunch, cant find info... Must search...

MrChromebox, 2023-10-15:
look on github and the brunchbook subreddit, brunch discord

Bobsisf, 2023-10-15:
Thank you for your time!

closed 2023-10-15 by Bobsisf