MrChromebox/scripts #324: HP Chromebook 13 G1: Script does nothing

MartinPlusPlus, 2023-10-25:
Greetings friends,

Part of my job is to wipe Chromebooks using these scripts. However with this particular model I am having some trouble.

https://github.com/MrChromebox/scripts/assets/25714534/d5389ba9-5f90-40d0-9987-035997a4ded3

^as you can see when I try to install legacy firmware the script doesn't do anything and doesn't even give an error. Lamentably I'm not to familiar with the debugging options for these scripts. But I'd needed I can provide more system information.

Any idea as to what might be causing this to occur?

ferrellsl, 2023-10-25:
You haven't disabled WP(write-protect).
-- The quality of your thoughts will determine the quality of your life. 

 
 

https://github.com/MrChromebox/scripts/assets/25714534/d5389ba9-5f90-40d0-9987-035997a4ded3

MrChromebox, 2023-10-25:
RW_LEGACY firmware is not supported for EOL devices; your terminal colors are causing the disabled/gray text to show up as white apparently.

Installing RW_LEGACY firmware doesn't have anything to do with wiping the device, so I'm curious what you are actually trying to do?

MartinPlusPlus, 2023-10-25:
I gotcha, I didn't know that these were EOL. Looks like I'll have to try another way then.

You're right, I'm just using the firmware to boot a USB to wipe it. I could have been more clear about that

MrChromebox, 2023-10-25:
> You're right, I'm just using the firmware to boot a USB to wipe it. I could have been more clear about that

and then what? A ChromeOS recovery USB will wipe it as well you know

MartinPlusPlus, 2023-10-26:
Yes, the USB image that the company uses also collects system information and uploads it to a server so that the Chromebook can be inventoried

MrChromebox, 2023-10-26:
I see. Is ChromeOS kept on the device or is Linux/Windows installed?

MrChromebox, 2024-10-14:
closing due to inactivity / lack of reply